rogerclarkmelbourne/Arduino_STM32 #803: Setting GPIO speed / Drive Strength / Slew Rate for F1

orkunkilinc, 2020-06-18:
Hi all,

To avoid EMC/EMI problems and glitches on the SPI slave I want to reduce drive current of the GPIO on the STM32F103ZCT6.

I can confirm that reducing slew rate definitely fixes the problem but the functions to change it are not implemented on this core. 

I basically want to change the correct register to reduce GPIO speed to 2MHZ with something like this. This is how it's done in STM32 Standart Peripheral Library.

 GPIO_InitStructure.GPIO_Speed = GPIO_Speed_2MHz;

Thank You.

stevstrong, 2020-06-21:
It is not clear how you set the output drive current on a SPI **slave** interface, because the slave only receives signals, and does not generate any.

For master SPI you can use the following function:
gpio_set_mode(GPIOx, Pyz, (GPIO_CR_CNF_OUTPUT_PP | GPIO_CR_MODE_OUTPUT_2MHZ));
wherein
- GPIOx = GPIOA for SPI1 and GPIOB for SPI2, and
- Pyz = the SPI pins for SPI1 (PA5/PA6/PA7) or SPI2 (PB13/PB14/PB15).

stevstrong, 2020-10-12:
Closed due to lack of activity from OP.

closed 2020-10-12 by stevstrong