rogerclarkmelbourne/Arduino_STM32 #805: multichannel example has a bug

afarhan, 2020-06-30:
The setup needs a startConversion to work:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/STM32ADC/examples/MultiChannelContinuousConversion/MultiChannelContinuousConversion.ino
 
  my_adc.setSampleRate(ADC_SMPR_1_5);//set the Sample Rate
  my_adc.setScanMode();              //set the ADC in Scan mode. 
  my_adc.setPins(pins, 2);           //set how many and which pins to convert.
  my_adc.setContinuous();            //set the ADC in continuous mode.

//set the DMA transfer for the ADC. 
//in this case we want to increment the memory side and run it in circular mode
//By doing this, we can read the last value sampled from the channels by reading the dataPoints array
  my_adc.setDMA(data_points, 2, (DMA_MINC_MODE | DMA_CIRC_MODE), NULL);
  my_adc.startConversion();  <<------------------------------- this is needed

stevstrong, 2020-06-30:
The example contains already the line:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/6d884b1f7712853901fb1c4507a8b4fd6b564c4a/STM32F1/libraries/STM32ADC/examples/MultiChannelContinuousConversion/MultiChannelContinuousConversion.ino#L53

You maybe use an older version?

stevstrong, 2020-10-12:
Closed due to inactivity.

closed 2020-10-12 by stevstrong