rogerclarkmelbourne/Arduino_STM32 #806: The upload-reset tool has wrong Linux 64-bit version

yulix, 2020-07-03:
In .../tools/linux64 directory,  use file command to check the version, it is 32-bit

yulix@01:~/stm32/Arduino_STM32/tools/linux64$ file upload-reset 
upload-reset: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=dc8dc5ad6de94179a9d534b1a34aeb6d906640b1, not stripped

In the same directory, the hid-flash  tool has 64-bit version.

yulix@01:~/stm32/Arduino_STM32/tools/linux64$ file hid-flash 
hid-flash: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=0781b11c35343e90d85375e5a2e0fdbc33104e1f, not stripped

stevstrong, 2020-10-12:
I am not familiar with LINUX.
What is the solution?

dewhisna, 2020-10-12:
@stevstrong -- this doesn't appear to be a "Linux issue" per se, but that whomever is building the `Arduino_STM32` bundle is apparently mixing 32-bit and 64-bit binaries in the tools folders.

stevstrong, 2023-11-03:
I really cannot understand and solve the issue.
Is there anyone who can help on this?

board707, 2025-07-01:
I using the Arduino_STM32 package on a Linux-64bit OS.

What I had to do to use a linux tools -
Change lines in platform.txt:
```
tools.maple_upload.path.linux={runtime.hardware.path}/tools/linux
tools.serial_upload.path.linux={runtime.hardware.path}/tools/linux
tools.stlink_upload.path.linux={runtime.hardware.path}/tools/linux
... etc ...
```
to
```
tools.maple_upload.path.linux={runtime.hardware.path}/tools/linux64
tools.serial_upload.path.linux={runtime.hardware.path}/tools/linux64
tools.stlink_upload.path.linux={runtime.hardware.path}/tools/linux64
```