rogerclarkmelbourne/Arduino_STM32 #808: Arduino_STM32 I2C scanner wire not working

Naveenkato, 2020-08-06:
Hy,
Im using MCP23017 to expand the GPIO,But here im using Wire.h library for communicate the STM32F103C8T6 with MCP23017.When i run the the i2c scanner program,it continuously returns no device found.But the softwire library detects the device present in I2C bus.

stevstrong, 2020-08-06:
Do you try to communicate on pins PB6/PB7 ?
Does the module have pull-up resistors on SDA/SCL lines?

dewhisna, 2020-08-06:
Possible duplicate of #799? (which I still haven't had time to look into since my lab isn't fully restored yet from the flooding)

If it is the same issue, the workaround in the meantime is to just use fixed/known device addresses and not use scanning to locate them.

stevstrong, 2020-08-07:
As @dewhisna suggested, try to use the module directly, not using the scanner.
Does the chip then work?

Naveenkato, 2020-08-10:
Yeah, the wire library working well with known devices.

stevstrong, 2020-12-07:
@Naveenkato , please test the latest version (see https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/1ac0186984e38de6fe1820afaf5b27cacae56d19), there was a fix for `Wire.h` regarding the STOP condition, it might solve this issue as well.

stevstrong, 2020-12-13:
No answer from OP, so I close this because it works now as expected.

closed 2020-12-13 by stevstrong