rogerclarkmelbourne/Arduino_STM32 #811: Crash over 64Kbytes

AntonioFromBrazil, 2020-08-18:
I am using an STM32F103CBT6 generic board, with 128K Flash and 20K RAM. My sketchs are being compiled with the Arduino IDE 1.8.13, using the Roger Clark's core. The compiller recognoize 20480 bytes of RAM (I am using only ~5K) and 131072 bytes as the maximum memory for programm but if the programm size go over 64k it crashs... It did not even start... It seams that the compiller is not managing address over 0x0800FFFF since the sketch crashs as soon as it sizes cross over that border. Bellow is a output issued at end of compillation (less than 64k). Is it possible to be a bug in the core ? Tks, Antonio
---------------------------------------------------------------------
http://stm32flash.googlecode.com/
Using Parser : Raw BINARY
Interface serial_w32: 115200 8E1
Version : 0x22
Option 1 : 0x00
Option 2 : 0x00
Device ID : 0x0410 (Medium-density)

    RAM : 20KiB (512b reserved by bootloader)
    Flash : 128KiB (sector size: 4x1024)
    Option RAM : 16b
    System RAM : 2KiB
    Write to memory
    Erasing memory
    Wrote address 0x0800fb38 (100.00%) Done.
    Starting execution at address 0x08000000... done.
---------------------------------------------------------------------

stevstrong, 2020-08-18:
Which upload method did you use?

AntonioFromBrazil, 2020-08-18:
Generally I use Serial but I have already tryied ST-Link and the result is the same. But it is curious that with ST-Link it reports only 64K !!! With Serial it repoprts 128K. The compiller output when I upload with ST-Link is:
--------------------------------------------------------------
ST-LINK SN: 3823040232124153354B4E00
ST-LINK Firmware version: V2J31S7 (Need Update)
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.2 V
Connection mode: Normal
Reset mode: Software reset
Device ID: 0x410 
Device flash Size: 64 Kbytes
Device family: STM32F10xx Medium-density
--------------------------------------------------------------

JiriBilek, 2020-08-18:
Are you really sure the MCU is STM32F103CBT6? Isn't it C8T6 instead?

AntonioFromBrazil, 2020-08-18:
It is a C8T6 but it has 128K according to ST Flash Load Demonstrator program... If I compile the skretch using the ST core it recognizes 128k but issues hundreds of error messages regarding to libraries... It is really confuse

JiriBilek, 2020-08-18:
Then try to flash some random 128K into the chip and verify it. Not every C8T6 has 128K.

stevstrong, 2020-08-18:
Make a simple sketch to demonstrate the issue, include a constant string of 50kB size to te sketch, and send it to serial.

AntonioFromBrazil, 2020-08-19:
ôoops.... Please accept my apologize. The board is a fake one and actually it does not have 128k but only 64k. If you tell to compiller your board have some amount of memory, it believes you but crashs after compilation... Thanks anyway

closed 2020-08-19 by stevstrong