rogerclarkmelbourne/Arduino_STM32 #813: Java Runtime PATH problem under Windows (EXCEPTION_ACCESS_VIOLATION)

dadul96, 2020-08-24:
This problem occurs when a second (different) java version is installed on a windows PC (besides the one provided with the Arduino IDE). Instead of using the provided java version of the Arduino IDE the `maple_upload.bat`-file uses the java version found in the environment path (oftentimes this is a different one).

This results in the following error message:
```
maple_loader v0.1
Resetting to bootloader via DTR pulse
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=12532, tid=0x000000000000127c
#
# JRE version: Java(TM) SE Runtime Environment (8.0_261-b12) (build 1.8.0_261-b12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.261-b12 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\Daniel\Documents\Arduino\hardware\Arduino_STM32\tools\win\hs_err_pid12532.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

To solve this problem the following line should be added to the maple_upload.bat:
`if exist "C:\Program Files (x86)\Arduino\java\bin" set PATH=C:\Program Files (x86)\Arduino\java\bin;%PATH%`

This line temporarily changes the PATH during the execution of this batch file and ensures that the correct java version is used.

This issue also has been discussed [here](https://www.stm32duino.com/viewtopic.php?t=604).

cad435, 2020-09-05:
Happens to me too, but for PlatformIO (wich afaik directly calls java with arguments without calling the .bat file).
Happened only after upgrading Java8 to newest release.

Solved by uninstalling Java8 and installing Java7 again... (Not recommended though, it is heavily outdated)

stevstrong, 2020-10-12:
I have to test the PR.

closed 2020-12-18 by dadul96

gibsonmb, 2023-01-29:
However, it's not fixed. I still get this error message (or very similar) using Arduino IDE 2.03 which of course does not in itself install or use JAVA. My installed version of JAVA is;
java version "1.8.0_361"
Java(TM) SE Runtime Environment (build 1.8.0_361-b09)
Java HotSpot(TM) 64-Bit Server VM (build 25.361-b09, mixed mode)

 and the error message I get when trying to compile BLINK is as follows;

Sketch uses 12804 bytes (9%) of program storage space. Maximum is 131072 bytes.
Global variables use 2064 bytes (10%) of dynamic memory, leaving 18416 bytes for local variables. Maximum is 20480 bytes.
maple_loader v0.1
Resetting to bootloader via DTR pulse
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=8016, tid=0x0000000000003b94
#
# JRE version: Java(TM) SE Runtime Environment (8.0_361) (build 1.8.0_361-b09)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.361-b09 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\Mike\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26\win\hs_err_pid8016.log
#
[hs_err_pid27268.log](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/10530233/hs_err_pid27268.log)

I am not smart enough to know how to fix this. It looks like a bug or incorrect entry in the platfrom.txt file maybe. Or is it something to do with using 64 bit Java.

stevstrong, 2023-11-03:
You should uninstall Java (JDK or JRE) for 64 bit platform and install 32 version instead.
This solved the issue for me.

superXnova21, 2025-04-03:
> You should uninstall Java (JDK or JRE) for 64 bit platform and install 32 version instead. This solved the issue for me.

You are the life saviour!!!! I got the same error and had been stuck for 2 days, tried 50-60 different methods including uninstalling ide and all that stuffs. And here at last the 32bit version worked. Thanks a lot my man.