rogerclarkmelbourne/Arduino_STM32 #815: no support serialEvent

chaohuazhu, 2020-08-31:
when support in main.c
int main(void)
{
	init();

	initVariant();

#if defined(USBCON)
	USBDevice.attach();
#endif
	
	setup();
    
	for (;;) {
		loop();
		if (serialEventRun) serialEventRun();
	}
        
	return 0;
}
when no support in main.c
int main(void) {
    setup();

    while (1) {
        loop();
    }
    return 0;
}

chaohuazhu, 2020-08-31:
or its fast enough to let it be, 72Mhz

stevstrong, 2020-12-17:
The serialEvent would be doable, but I wonder whether there is a real demand to implement that.

rogerclarkmelbourne, 2021-03-06:
Unless things have changed recently... 

The SerialEvent is not actually coded as a real event in the underlying core code.

The hidden main loop simply checks for data in the Serial buffer, hence most people just implement this in their own code

rogerclarkmelbourne, 2021-03-06:
I'm going to close this thread, because the OP posted over 6 months ago, and didn't respond to the reply from @stevstrong

closed 2021-03-06 by rogerclarkmelbourne