rogerclarkmelbourne/Arduino_STM32 #816: stm32core and  arduino ide with gsm

laxman432, 2020-08-31:
http://dan.drown.org/stm32duino/package_STM32duino_index.json

hi we used above link foraccesing stm32cores we done software 2017.9.21 core recently it is updated now with new cores gsm not working i am using pa2,pa3 as serial1 it is executing but not getting any messages when i check with old binary it is working can u help me in that it is very urgent for me

stevstrong, 2020-08-31:
Please provide more information about what exactly is not working, and post a simple example which shows the problem case.

laxman432, 2020-08-31:
i am not able to recieve sms from my hardware here we
used arduino 1.8.13 latest stm32 core when i check with previas versions
2017 .9.21 it was working now that 2017 lbrary not able to put it we need
its very urgent help me in this

*                                  Thanks & Regards,*
                                  * M.P.LAXMAN M.tech (Ph.D.)*
*                            CONTACT: **8143710109 *

fpistm, 2020-08-31:
Hi @stevstrong 
FYI: https://github.com/stm32duino/Arduino_Core_STM32/issues/1158

stevstrong, 2020-08-31:
Serial 1 uses PA9/PA10 as Tx/Rx.
Serial 2 uses PA2/PA3 as Tx/Rx.
It seems that you are mixing Serial1 with Serial2.
Try to use Serial2 in your code or change the pins to PA9/PA10.

laxman432, 2020-08-31:
Is there any changes from 2017.9.21 to new core I have old binary in that we done using serial1 only still it is working when I compile same code with updated core it is not communicating with gsm

stevstrong, 2020-08-31:
There have been a lot of changes in the core since 2017.09.21.
You need to use `Serial2`.

laxman432, 2020-08-31:
If I want use old version as my binary working perfect how to setup that can u give that stm32 core.actually we need it very urgent we don't have time to resear h on it why because we are in agriculture field my customers are asking

stevstrong, 2020-08-31:
If the old binary works, use that.
Otherwise you have to change the code as suggested above.

**Please be aware that for this project there is no prefessional support, the collborators are investing their own free time.**

laxman432, 2020-08-31:
How to download that 2017.9.21  version  can u give any link of that we tried but not got proper one and after downloading how to access that can u give me suggestions and links

stevstrong, 2020-08-31:
Go this link: https://github.com/rogerclarkmelbourne/Arduino_STM32/commits/master
and browse the code for the desired date.
Then press the button "Code".
Exptract the files under the already existing directory (overwrite).
Also see: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

laxman432, 2020-09-01:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

Asper above link under preferences in Arduino ide no need to give any json file link right ? If need to give which link need to give ?

rogerclarkmelbourne, 2020-09-01:
@laxman432 

The JSON installation files are maintained by @ddrown , and are not part of the files in this repo.

For direct installation of the core files, see the wiki

See https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

Except download the zip for the version you want, as explained by @stevstrong 

You also need to remove the version you installed by JSON.  Post to the Arduino.cc forum if you don't know how to do that, as the Arduino IDE is also a separate thing from this repository.

laxman432, 2020-09-01:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

WE DONE ACCORDING TO THAT LINK BUT WE ARE NOT ABLE TO SEE STM32 BOARDS IN BOARDS MANGER IN ARDUINO

laxman432, 2020-09-01:
unknown protocol: c
unknown protocol: c
Could not find boards.txt in C:\Users\ANU\Documents\Arduino\hardware\Arduino_STM32-master\Arduino_STM32-master. Is it pre-1.5?
WARNING: Error loading hardware folder C:\Users\ANU\Documents\Arduino\hardware\Arduino_STM32-master
  No valid hardware definitions found in folder Arduino_STM32-master.
Could not find boards.txt in C:\Users\ANU\Documents\Arduino\hardware\Arduino_STM32-master\Arduino_STM32-master. Is it pre-1.5?
WARNING: Error loading hardware folder C:\Users\ANU\Documents\Arduino\hardware\Arduino_STM32-master
  No valid hardware definitions found in folder Arduino_STM32-master.

stevstrong, 2020-09-01:
Check your local files, the extraction went probably wrong.
Alternaatively, you can try to get the old version of the previously instaled core file via the board manager (JSON file).

laxman432, 2020-09-01:
how to get that old versions we are not able find that we are trying to get
it but no links to do like that can u give any links

*                                  Thanks & Regards,*
                                  * M.P.LAXMAN M.tech (Ph.D.)*
*                            CONTACT: **8143710109 *

rogerclarkmelbourne, 2020-09-01:
@laxman432 

I don't know if @ddrown  keeps old versions. You could try www.archive.org
Or you could contact @ddrown somehow. (I don't have his contact details)

stevstrong, 2020-09-01:
Here the link to the version dated 2017.09.21 : https://github.com/rogerclarkmelbourne/Arduino_STM32/archive/f8d28b8663a53fa37f75c2f20ed0aebc81c83fba.zip

laxman432, 2020-09-01:
how to link that version with arduino ide please give me procedure 

previasly we installed that using boardmanager 

now here we got zip file

stevstrong, 2020-09-01:
I have already provided the link to the installation wiki page: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

stevstrong, 2020-10-12:
@laxman432, is there any update on this?

stevstrong, 2020-11-14:
Close due to inactivity.

closed 2020-11-14 by stevstrong