rogerclarkmelbourne/Arduino_STM32 #817: STLINK Programmer not Available

Avong, 2020-08-31:
Hello Dear, 

I downloaded, extrated, renamed to STM32 and then placed the folder inside hardware folder. 
but any board i selected, the programmer selection indicates " No programmer available for this Board" 

When i pressed "upload" button, i get the message below but no file uploaded. 

My laptop runs Ubuntu 20.20 64bits.
Arduino is installed inside /opt folder. 

**dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Waiting for /dev/ttyACM0 serial...Done**

stevstrong, 2020-10-12:
I am not familiar with LINUX.
Who can help on this?

dewhisna, 2020-10-12:
@Avong - Did you set the udev rules so that it can map the USB Vendor ID and Product ID to be a `ttyACM` device?  These rules files will be in `/etc/udev/rules.d/`.  Google the details on how to edit these (and there are various scripts floating around out there that will automate it for you), but it will be something like:

```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="374f", MODE:="0666"
KERNEL=="ttyACM*", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="374f", MODE:="0666"
```

And I think some people run into issues with the ModemManager trying to probe it first, causing the software to timeout searching for the device.  So you might want to add something like this to disable it:

```
ATTRS{idVendor}=="0483", ATTRS{idProduct}=="374f", ENV{ID_MM_DEVICE_IGNORE}="1"
```

For the specific hex numbers above, you'll want to run `lsusb` to list the UBS IDs found to make sure it matches your device.  There's about a half-dozen different Device IDs for the ST-Link, depending on its mode.  I used `0483` and `374f` in the above examples, which I think is ST-Link V3 in Normal mode.  I have the actual ST installation on my computer, so I have all of the identifiers, but I don't know which one matches your device, so use `lsusb` to list and figure out what you have.

You'll also need to check your user account and make sure you are member of the `dialout` group.  Run `groups` to list what groups you are a member of and if `dialout` isn't listed, Google how to add it...

I think once you add udev rules and make sure your user account is in `dialout`, it should work.

Avong, 2020-10-12:
> @Avong - Did you set the udev rules so that it can map the USB Vendor ID and Product ID to be a `ttyACM` device? These rules files will be in `/etc/udev/rules.d/`. Google the details on how to edit [...]

Thanks for the help. 
I was able to solve that:

What i did:
Uninstalled the arduino IDE
Clean all arduino related files. 
Reinstalled Arduino IDE, 
Reinstalled STM32 support file, this time, i used the 
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

Now everything work just fine. 

Thank once more.

closed 2020-10-12 by stevstrong