rogerclarkmelbourne/Arduino_STM32 #821: Installing Drivers - Access Is Denied

NathanielJS1541, 2020-10-07:
Whenever I run `install_drivers.bat` I get the following error:
[image]
I have tried running the .bat file as an administrator, and I have tried running it from an administrator command prompt but I get the same error each time.

Any help would be appreciated.

stevstrong, 2020-10-07:
Try to temporarily deactivate any antivirus software you have running.
Btw, the core files should be installed under <user>/Documents/Arduino/hardware/, see installation instructions in Wiki.

NathanielJS1541, 2020-10-07:
Thanks for the help - It was an issue with Kaspersky. My files are moved to the correct place now too - I was following a guide online and those steps weren't included - Thanks a lot!

closed 2020-10-07 by NathanielJS1541