rogerclarkmelbourne/Arduino_STM32 #823: SPI in async mode will leave IRQ enabled

mean00, 2020-10-27:
When using the SPI IRQ in async mode, the SPI IRQ will be left enabled
The dma_detach_function will not disable it, with comment  /* Don't use nvic_irq_disable()! Think about DMA2 channels 4 and 5. */
The problem is when alternating between blocking SPI send and async SPI Send, the irq is left enabled and non managed interrupt will happen. Additionnaly if that happens the dma_irq_handler will loop as it does not clear the interrupt if no handler is present.

Crude hack is to do that in the SPI itself , i.e. in  SPIClass::onTransmit & SPIClass::EventCallback
i.e. explicitely call nvic_irq_disable.

It works, but is there a better way ? 

(The same problem probably exists with transfer, but i'm only dealing with transmit)

stevstrong, 2020-10-27:
Which device? F1 or F4?
Do you have a simple example sketch which shows the problem?

mean00, 2020-10-27:
Hi
I'm using the  F1 (f103 actually)
i can try creating a simple use case, it might take a bit of time as  i'm using a very different environment than the arduino one, with a lot of dependencies.

But the gist of it is : 
* Do a spi.dmaSendAsync()
* Do a spi.dmaSend()

You'll run into the 1st problem, i.e. it loops in dma_irq_handler when the spi.dmaSend() is over because the nvic irq is still enabled.
The asymetry in dma_detach_interrupt / dma_attach_interrupt is the root cause i believe

mean00, 2020-10-28:
Hi
Please find attached a tar ball with 2 files : 
- The demo script. It will loop inside dma_irq_handler.
- My current workaround, sort of of fixes the above + trigger a debugger exception if there is a spurious interrupt

Thank you

[spiTest.tar.gz](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/5450053/spiTest.tar.gz)

mean00, 2020-12-05:
ping ?

stevstrong, 2020-12-05:
I just committed a fix for this, I have tested and it works correctly now for me.
Please check it and reopen the issue if the problem still persists.

closed 2020-12-05 by stevstrong

mean00, 2020-12-05:
seems to work fine
Thanks a lot!

commit 278c520d (2020-12-05): F1: fix #823 - DMA IRQ related

- manage IRQ enable/disable for DMA2 channels 4 and 5 which share a common IRQ line
- clear ISR bits for DMA IRQ even if no handler is set