rogerclarkmelbourne/Arduino_STM32 #824: ADC thereshold detect & interrupt

AntonioFromBrazil, 2020-10-28:
I am trying to generate an interrupt when limits defined by setAnalogWatchdog(channel, HighLimit, LowLimit) are crossed but I think that function is not working well since no interrupts are generated. Maybe I forgot something... I would suggest an example of how to work with analog watchdog...

stevstrong, 2020-10-28:
Can you post a simple sketch you have tried?

AntonioFromBrazil, 2020-10-28:
Sure... thanks stevtrong. I tried many different variations over this attached example but no way !!! I will appreciate any help.

`#include <STM32ADC.h>
STM32ADC myADC(ADC1); 

uint8   ana   =PA0;  // analogic input pin

void setup() {
  Serial1.begin(115200);
  pinMode(PA0, INPUT_ANALOG);
  myADC.calibrate();
  myADC.setSampleRate(ADC_SMPR_1_5);
  myADC.setPins(&ana, 1);
  myADC.setAnalogWatchdog(0, 2100, 1900); //high low limit at channel 0 (PA0)
  myADC.attachAnalogWatchdogInterrupt(adc_int);
  myADC.attachInterrupt(adc_int, ADC_AWD);
  myADC.setContinuous();
  myADC.startConversion();
}

void adc_int()
{
  Serial.println("int");
}

void loop() {
}`

AntonioFromBrazil, 2020-10-28:
in the example there is no difference if I use myADC.attachAnalogWatchdogInterrupt(adc_int);
or myADC.attachInterrupt(adc_int, ADC_AWD); or both...

dewhisna, 2020-10-28:
@AntonioFromBrazil -- you do realize that you aren't using the same serial port in `setup()` and `adc_int()`, right?  A quick grep through the framework source shows that `Serial` is `USBSerial` and `Serial1` is `HardwareSerial` (i.e. one of the USARTs).

So what port are you monitoring?

AntonioFromBrazil, 2020-10-28:
Serial1. There is only a typewriter mistake on example...

stevstrong, 2020-10-28:
Your application has two errors.

First, if you start continues conversion, then the ADC will convert in loop.
This means that the interrupt comes regularly in very short intervals and will be nested because the ADC ISR cannot run faster than the ADC conversion time. After some time the CPU will simply hang, this is why you don't see anything on the serial monitor.

The workaround to this is to disable the conversion or the watchdog alone in the ISR, and restart it in the main loop. Or better use single conversion and restart it each time in the main loop after you printed it on the serial monitor. PLease check the single interrupt example.

Second, the AWD interrupt is not yet implemented in the lib. Although the functions are there, but behind them the IRQ will not be enabled.

I am currently making some amendments on ADC lib to solve these issues, I hope I will finish it today or tomorrow, I will then post here.

AntonioFromBrazil, 2020-10-28:
Woow... it is great ! Do you thik would be a good idea to have one awd code example as part of this library ?

stevstrong, 2020-10-28:
Yes, I will make one.

AntonioFromBrazil, 2020-10-30:
stevstrong, If there is anyway I can help in that amendments you have mentioned, just tell me. I am available to work on it if you give me some directions. Thanks

stevstrong, 2020-10-30:
I need to commit some changes in the core files.

stevstrong, 2020-11-01:
I just committed some [changes in the lib](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/b6f73e96a6b81a3c894119339cf06f2d8bdc0c42), and added an example how to handle the AWD.
Please check it out and send me your comments.

AntonioFromBrazil, 2020-11-01:
I did not find any examples of AWD usage... Please tell me where the example you have added  is located. Thanks

stevstrong, 2020-11-01:
Ooops, I forgot to added to the commit list.
Now you can find it here: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/STM32ADC/examples/AnalogWatchdogInterrupt/AnalogWatchdogInterrupt.ino

AntonioFromBrazil, 2020-11-01:
It works perfectly... I think that example will help many others. Thanks a lot !!!

stevstrong, 2020-11-14:
Issue solved.

closed 2020-11-14 by stevstrong