rogerclarkmelbourne/Arduino_STM32 #832: stm32 arduino

prashanthchinnu, 2020-12-10:
hi sir... this is prashanth.,
im uisng this links..
 http://dan.drown.org/stm32duino/package_STM32duino_index.json     and
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
but  Serial2 is not communicate.. it shows error.
plz can u help me

rogerclarkmelbourne, 2020-12-10:
https://github.com/stm32duino is a completely different Arduino Core. Its the official STM Arduino core.

You may have conflicts installing 2 completely different Arduino Cores for the STM32

Code will not work the same on both cores.

This is not a support forum.  Try posting to  https://www.stm32duino.com

prashanthchinnu, 2020-12-10:
thank you for your reply sir...
which link is better to use sir for serial2.

rogerclarkmelbourne, 2020-12-10:
>which link is better to use sir for serial2. plz can you send me that link

This is not a support forum and you are not posting a bug. 

Please try 

https://www.stm32duino.com

closed 2020-12-10 by rogerclarkmelbourne

prashanthchinnu, 2020-12-10:
ok sir thank you..

prashanthchinnu, 2020-12-10:
it is showing like this sir.. by using those links.,
Error compiling for board Generic STM32F1 series.