rogerclarkmelbourne/Arduino_STM32 #834: A fatal error has been detected by the Java Runtime Environment

Proxcentaur, 2020-12-14:
I have been using STM32F1C series since 5 days ago. For the first time using this board, I have successfully upload a blink program both of using FTDI and STM32 bootloader through USB port. But today I got a strange error like this:
**
maple_loader v0.1
Resetting to bootloader via DTR pulse
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=5852, tid=1516
#
# JRE version: Java(TM) SE Runtime Environment (15.0.1+9) (build 15.0.1+9-18)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (15.0.1+9-18, mixed mode, sharing, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\dartmedia\Documents\Arduino\hardware\Arduino_STM32\tools\win\hs_err_pid5852.log
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.**

Why did this happen? I have updated STM32 boards in Arduino IDE and my JDK to the last version but the error still occurs. 
What should I do ?

rogerclarkmelbourne, 2020-12-14:
Github issues are not a support forum.

You should post this to 

https://www.stm32duino.com/

stevstrong, 2020-12-15:
This seems to be a Java or Arduino IDE issue, not core issue. So I will close this.

stevstrong, 2020-12-15:
Possible duplicate of #813

closed 2020-12-15 by stevstrong

rogerclarkmelbourne, 2020-12-15:
Yes. Also the OP didn't reply

stevstrong, 2020-12-15:
As a possible solution you may try to execute the following line in a CMD window (with admin privileges):
`set PATH=C:\Program Files (x86)\Arduino\java\bin;%PATH%`

Proxcentaur, 2020-12-15:
I have solved this issue by pressing the STM32 reset button during uploading the program

ArminJo, 2023-08-01:
The reason is this line in maple_upload.bat:  `java -jar maple_loader.jar %1 %2 %3 %4`
The java.exe is found by using the Windows %PATH% pointing to your java version15 which is incompatible with maple_loader.jar.
Thefore the `java `must be replaced by `......\arduino\java\bin\java` which points to the Arduino Java 1.8!
The arduino internal usage of `{the runtime.ide.path}/java/bin` does not work 😞 .

cino893, 2025-03-27:
@ArminJo 
Sheesh this fix runnin’ smooth af, frfr, no bugs no cap, lovu man. 

Using genuine STM32 board lib I was removing serial USB bootloader everytime i uploaded my code and had to UART bootloader all the time and compilation took longer. Now everything work seamlessly.

In order to solve this issue you have to(W11):
1. Download and install 7zip: https://7-zip.org.pl
2. Go to `C:\Users\marcin\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26` or similar path where `maple_upload.bat` is located
3. Download Legacy java 1.8 `https://repo.huaweicloud.com/java/jdk/8u202-b08/jdk-8u202-windows-x64.exe` or from official Oracle but you have to have account.
4. Download, extract, and run java_install to portable script like ` .\unpack-jdk-x64.bat .\jdk-8u202-windows-x64.exe`
`https://gist.github.com/cino893/1a542ad1cefb0fbd56c506a350679c42` where you can place your path to 7zip
5. AAAAAAAAnd last edit your `maple_upload.bat`  Ln 8 `set PATH=%5\java\bin;%PATH%` to in mine case `set PATH=C:\Users\marcin\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26\win\jdk\jre\bin;%PATH%`

[image]