rogerclarkmelbourne/Arduino_STM32 #838: You wordpress blog

Anton2k, 2021-01-08:
Hi roger I hope you don't mind me commenting on one of your projects like this but, I can't seem to comment on your blog to ask a question, specifically this post https://www.rogerclark.net/ttgo-t5-eink-display-update/ but I also can't comment on any blog posts.

Was hoping you would be able to answer a question relating to the use of batteries on the TTGO T5 (my version is 2.3 not sure what you are using in your post.)

Basically I was wanting to know / trying to figure out if the TTGO T5 had undervolt protection built in, as far as I can make out it does have overvolt protection based on my initial LiPo tests (won't go above 4.2v). I seen that you done some modifications using the TP4055, I actually have a few TP4056's laying around. But yea my main question is, if I was to use a LiPo or Ion wired into the battery connector on the back of the board, will it kill the batteries over time? I'm pretty sure it won't over charge them, but not sure about the discharge, I want to try and do all I need to with the TTGO T5 without to many external modifications. In my project I am going to try and get it into ULP mode so it can still wake it's self on a timer, do some stuff via wifi then go back to sleep, will be on for 1 mins twice a day, so 2 mins total. I want to try and get a couple of months run time out of it.

Thanks and apologies for commenting on an issues, feel free to delete or I could sent you my email / comment on your blog once it is fixed.

Anton2k, 2021-01-23:
???

rogerclarkmelbourne, 2021-01-24:
This is the issues list for the Arduino STM32 core.
**Its NOT general message system to me.**

I have no idea about your question

closed 2021-01-24 by rogerclarkmelbourne

Anton2k, 2021-01-29:
Basically I can't post a comment on your blog, and have no other way that I can see of contacting you to make you aware of it, just thought you might like to know.