rogerclarkmelbourne/Arduino_STM32 #839: Can't use ArduinoModbus with STM32F1 Blue Pill (Maple Mini clone)

requeijaum, 2021-01-15:
Seems like a bug in this core. I already opened issues in other repositories.

https://github.com/stm32duino/Arduino_Core_STM32/issues/1279

https://github.com/arduino-libraries/ArduinoModbus/issues/47

stevstrong, 2021-01-23:
It seems that this core is not prepared to include <Arduino.h> in a C file.
The solution is to make conditional the include of CPP header files only when compiling with g++.

In order to get rid of the errors, please try to replace the attached files in the directory `\Arduino_STM32\STM32F1\cores\maple`.
[cores-maple-include-files.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/5860290/cores-maple-include-files.zip)
I have tested and it works for me but please let me know if this resolves the issue for you, too.

stevstrong, 2021-02-07:
Have you managed to try my suggestion?

stevstrong, 2021-02-21:
[This commit](https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/5175e3b728096231ae3758a4bb66077e6249a5fe) should solve the problem.

closed 2021-02-21 by stevstrong