rogerclarkmelbourne/Arduino_STM32 #840: STM32F103CB STLink V2 Arduino 1.8.13 MAC stops upload Coreid: 2ba01477

pirvcb, 2021-01-28:
STM32F103CB STLink V2 OSX stops upload

Post by PIRVCB » Thu Jan 28, 2021 12:30 am
Hello there!
Got a STM32F103CB board. In Arduino IDE 1.8.13 running on WIN7 all goes ok.
Tried 2 different STLink V2 chineese clones. Same results.
Arduino Log:
STM32 ST-LINK CLI v3.3.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: 48FF6E065188515532321067
ST-LINK Firmware version: V2J37S7 (Need Update)
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.5 V
Connection mode: Normal
Reset mode: Software reset
Device ID: 0x410
Device flash Size: 128 Kbytes
Device family: STM32F10xx Medium-density

Loading file...
Flash Programming:
File : C:\Users\kkk\AppData\Local\Temp\arduino_build_5217\V1V2V3V4_LPFNOVARIABLESTATE_STM.ino.bin
Address : 0x08000000
Memory programming...
0%�������������� 29%��������������� 59%��������������� 89%������ 100%
Memory programmed in 3s and 385ms.
Programming Complete.

MCU Reset.

Application started.

STCube LOG:
23:38:28 : Voltage : 3.19V
23:38:28 : SWD freq : 4000 KHz
23:38:28 : Connect mode: Normal
23:38:28 : Reset mode : Software reset
23:38:28 : Device ID : 0x410
23:38:28 : Revision ID : Rev X
23:38:29 : UPLOADING OPTION BYTES DATA ...
23:38:29 : Bank : 0x00
23:38:29 : Address : 0x4002201c
23:38:29 : Size : 8 Bytes
23:38:29 : Bank : 0x01
23:38:29 : Address : 0x1ffff800
23:38:29 : Size : 16 Bytes
23:38:29 : UPLOADING ...
23:38:29 : Size : 1024 Bytes
23:38:29 : Address : 0x8000000
23:38:29 : Read progress:
23:38:29 : Data read successfully
23:38:29 : Time elapsed during the read operation is: 00:00:00.015

PROBLEM IN MAC OSX 10.10.5

When move to OSX 10.10.5, with same version of Arduino IDE 1.8.13 and same STM32 library Roger Clark , the latest, i go in the following issue.
Memory gets erased but then STLink stops before upload the new memory content.

LOG:

Arduino:1.8.13 (Mac OS X), Scheda:"Generic STM32F103C series, STM32F103CB (20k RAM. 128k Flash), STLink, 72Mhz (Normal), Fastest (-O3)"

Lo sketch usa 21348 byte (16%) dello spazio disponibile per i programmi. Il massimo è 131072 byte.
Le variabili globali usano 4432 byte (21%) di memoria dinamica, lasciando altri 16048 byte liberi per le variabili locali. Il massimo è 20480 byte.
2021-01-27T23:46:38 INFO /Users/kuwatay/src/stlink-master/src/usb.c: -- exit_dfu_mode
2021-01-27T23:46:38 INFO /Users/kuwatay/src/stlink-master/src/common.c: Loading device parameters....
2021-01-27T23:46:38 INFO /Users/kuwatay/src/stlink-master/src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-01-27T23:46:38 INFO /Users/kuwatay/src/stlink-master/src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2021-01-27T23:46:38 INFO /Users/kuwatay/src/stlink-master/src/common.c: Attempting to write 23396 (0x5b64) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08005400 erased2021-01-27T23:46:39 INFO /Users/kuwatay/src/stlink-master/src/common.c: Finished erasing 23 pages of 1024 (0x400) bytes
2021-01-27T23:46:39 INFO /Users/kuwatay/src/stlink-master/src/common.c: Starting Flash write for VL/F0/F3 core id
2021-01-27T23:46:39 ERROR /Users/kuwatay/src/stlink-master/src/flash_loader.c: unknown coreid, not sure what flash loader to use, aborting! coreid: 2ba01477, chipid: 410
2021-01-27T23:46:39 WARN /Users/kuwatay/src/stlink-master/src/flash_loader.c: Failed to write flash loader to sram!
2021-01-27T23:46:39 ERROR /Users/kuwatay/src/stlink-master/src/common.c: stlink_flash_loader_init() == -1
2021-01-27T23:46:39 ERROR /Users/kuwatay/src/stlink-master/src/common.c: stlink_flash_loader_init() == -1

Flash page at addr: 0x08005800 erased
stlink_fwrite_flash() == -1

Other generic boards, like Blue Pills 103xx got already programmed without errors with same setup.

The board pictures are below.

https://drive.google.com/file/d/1MdYmImp78xfoDGcAevWeDqT7j-b2_0ew/view?usp=sharing
https://drive.google.com/file/d/1eML3qYDpD0A5FfYI5h58dbSnmTH40FvT/view?usp=sharing

TIA!!!
Davide

tonaalt, 2021-03-06:
I am having same error, I think you have clone, not real stm32.
I think this was solved in stlink repo, need to update one in this repo.
https://github.com/stlink-org/stlink/pull/834/commits/8562e9c7effb3f297bd1420b4ce2237cb751472a

tonaalt, 2021-03-06:
This is the chip that I have APM32F103CB 128KB MCU, and it gives same error in linux. Robotdyn black board.

rogerclarkmelbourne, 2021-03-06:
You both may have fake chips. 

This is a very widespread problem.

The official STM software will not upload via STLink to the fake chips

tonaalt, 2021-03-06:
Not fake, a clone as I stated, it is APM32F103CB. and that also reads on chip so not fake.
pirvcb  might have a fake, that I don't know.
And it uploads with no problem with other core and cubeProgrammer(SWD), with STLink.
And like I said that id is added to stlink repo linked above also.

rogerclarkmelbourne, 2021-03-07:
@tonaalt 

AFIK. This core either uses the official STM binary for STLinkCLI  , or uses Texane STLink.

The link you posted (stlink-org/stlink@8562e9c)  seems to be to a different STLink implementation entirely

closed 2021-06-08 by pirvcb

kokospalme, 2022-04-08:
my ST-link v2 clone doesn't even show up when I try to choose it as programmer... 
I downloaded the STM8 core from tenbaht (https://github.com/tenbaht/sduino) --> St-Link V2 shows up and I can upload my code withput problems.

Can this be combined somehow to get the st-link v2 clones work for the stm32 core? This would be great...