rogerclarkmelbourne/Arduino_STM32 #842: PWM_OPEN_DRAIN not open

suuuehgi, 2021-02-03:
Hey,

I connected 5 V -> 1K -> Pin PB9 with the following code

```C
const int pin_pwm = PB9;
pinMode(pin_pwm, PWM_OPEN_DRAIN);
pwmWrite(pin_pwm, 20000);
```

and I get a PWM signal at the pin of slightly above 3.3 V, not 5 V.
So open is not really open but rather around 2 K. :/

stevstrong, 2021-02-03:
How do you measure the output voltage? With a volt-meter? You should check it with an oscilloscope.

suuuehgi, 2021-02-03:
I used an Oscilloscope.

suuuehgi, 2021-02-03:
Thank you very much for the investigation so far! Just in case I miss something obvious, here's the minimal setup:

[image]

```c
const int pin_pwm = PB9;

void setup() {
    pinMode(pin_pwm, PWM_OPEN_DRAIN);
    pwmWrite(pin_pwm, 20000);
}

void loop() {
}
```
[image]
[image]

stevstrong, 2021-02-03:
I tested with exactly the same HW setup, and here is my plot:

[image]

Try with another resistor and/or board.

suuuehgi, 2021-02-03:
Same on another board from the same order. I have one last from the other batch I'm gonna test now ...

suuuehgi, 2021-02-03:
The one from the old batch gives 5 V. -.-

JiriBilek, 2021-02-03:
Just curious, are you sure your CPU is a genuine one? There is a lot of clones/counterfeits on the market.

suuuehgi, 2021-02-03:
> Just curious, are you sure your CPU is a genuine one? There is a lot of clones/counterfeits on the market.

Well, I don't know. For the sake of ST, I guess it's fake.

[image]

Tested now pin PB6-9 + PA8-10. 3.3 V PWM on all of them.

JiriBilek, 2021-02-03:
The CPU in the picture looks good.

suuuehgi, 2021-02-03:
I think I can close this as it seems to be μC-related. Thank you both for your efforts!

closed 2021-02-03 by suuuehgi

suuuehgi, 2021-02-03:
I got it now working in software toggling between `pinMode` `OUTPUT, LOW` and `INPUT`.

stevstrong, 2021-02-04:
Can you please share how exactly you did it?

suuuehgi, 2021-02-04:
> Can you please share how exactly you did it?

Yeah, sure!

```c
#define PWM_PERIOD 313    // in microseconds / 1.6 --> ~ 2000 Hz

const int pin_pwm = PB9;

HardwareTimer timer(1);

void setup() {

    pinMode(pin_pwm, OUTPUT_OPEN_DRAIN);

    timer.pause();
    timer.setPeriod(PWM_PERIOD);
    timer.setMode(TIMER_CH1, TIMER_OUTPUT_COMPARE);

    // Fire interrupt at 50 % of the counter
    timer.setCompare(TIMER_CH1, timer.getOverflow()/2);
    timer.attachInterrupt(TIMER_CH1, handler_pwm_high);

    // Channel 0 is for overflow interrupt (update interrupt)
    timer.attachInterrupt(0, handler_pwm_low);

    timer.refresh();
    timer.resume();

}

void loop() {
}

void handler_pwm_high(void) {
    digitalWrite(pin_pwm, HIGH);
}
void handler_pwm_low(void) {
    digitalWrite(pin_pwm, LOW);
}
```

Some additional notes:

 - I noticed that `OUTPUT_OPEN_DRAIN` works as intended and just `PWM_OPEN_DRAIN` is broken. I switched to that and updated my comment above accordingly.
 - `PWM_PERIOD` is -- why ever -- not directly the inverse of the frequency but scaled by a constant factor of 1.6. Hence 500 resulted not in 2000 Hz but in 1250 Hz. Thus I used 500 / 1.6 here.
 - Channel 0 behavior of the `attachInterrupt` method is not documented but in a source code comment.

stevstrong, 2021-02-04:
I suppose that this workaround you apply only for the "fake" MCU.

This code generates the correct timing and level:
```
#define PWM_PIN		PB9 // Timer4 Ch 4
#define PWM_PERIOD	500 // in µs, 2000 HZ

#define myTimer Timer4

void setup()
{
    pinMode(PWM_PIN, PWM_OPEN_DRAIN);

    myTimer.pause();
    myTimer.setPeriod(PWM_PERIOD);
    myTimer.setCompare(TIMER_CH4, myTimer.getOverflow()/2); // set duty cycle
    myTimer.refresh();
    myTimer.resume();
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

If the PWM period is not correct, it can be another sign for a fake MCU, or defective MCU/board.

suuuehgi, 2021-02-04:
> I suppose that this workaround you apply only for the "fake" MCU.

Yes, certainly. Otherwise I wouldn't have to use interrupts and could just let it run "in hardware".

Thank you for the double check and the hint that this might be another deviation from the original. 😒

It might be coincidence ... but I noticed that 74 MHz / 45 MHz is about 1.6. Might it be that the μC runs at 45 MHz?

stevstrong, 2021-02-04:
Not sure about the deviation, but normally it runs with 72MHz. Maybe 72/48? That would be 1.5. Have you checked the onboard crystal?

suuuehgi, 2021-02-04:
Ah, right, I mistook it with 48 MHz. Then it's probably meaningless. The onboard crystal is the std 8 MHz.