rogerclarkmelbourne/Arduino_STM32 #845: RTClock.h: No such file or directory

nesergen, 2021-02-10:
I am trying to compile the BluePill-RTClock-test sketch for stm32f103. But I get an error.

```
Alternatives for RTClock.h: []
ResolveLibrary(RTClock.h)
  -> candidates: []
BluePill-RTClock-test:27:10: fatal error: RTClock.h: No such file or directory
   27 | #include <RTClock.h>
      |          ^~~~~~~~~~~
compilation terminated.
exit status 1
RTClock.h: No such file or directory
```
What could be the problem?

stevstrong, 2021-02-10:
Which version of Arduino IDE are you using?
Which board have you selected?
Which upload method?

nesergen, 2021-02-11:
> Which version of Arduino IDE are you using?

IDE 1.8.14

> Which board have you selected?

Generic STM32F1 series -> BluePill F103C8

> Which upload method?

Maple DFU Bootloader 2.0.

I am not trying to upload a sketch to the board, just checking that the code is correct.

stevstrong, 2021-02-11:
Well,  I tried and the same example works for me, no error message, that is why I asked those questions.
Btw, there is no BluepilF103 selection option for this core, maybe you have used the official STM core?
Please post an image of the board selection options of Arduino.

nesergen, 2021-02-11:
oh, you're right. I installed the stm32duino / Arduino_Core_STM32 version. The presence of different branches and libraries confuses me. It is a pity that the versions have a library incompatibility.
Can I install your version in parallel?

stevstrong, 2021-02-11:
Yes, it is possible to use both cores. You only have to pay attention which board you select, with that selection you also select the underlying core (show in the top of the board lists).

closed 2021-02-11 by stevstrong