rogerclarkmelbourne/Arduino_STM32 #847: mapple driver not detected in mac

ehrenmann1977, 2021-02-27:
uploading the generic firmware with pc13 to blue pill then connecting the micro usb to a mac, it does not detect the device on the serial port even after rebooting. Lsusb does not show it. Is there a mapple driver for mac?

rogerclarkmelbourne, 2021-02-27:
No. There is no specific driver for OSX.

The bootloader   ( https://github.com/rogerclarkmelbourne/STM32duino-bootloader )  is a generic USB CDCACM device.

However, it looks like more recent versions of MacOS may no longer support this generic device, for either software or hardware reasons.

I suggest you post to 

https://www.stm32duino.com

stevstrong, 2021-02-27:
I think this is rather bootloader related and not core related.
There is a `binaries` and `bootloader_only_binaires` folder under the link mentioned by Roger, the first has USB serial part included, the other not.
Which one are you using?

rogerclarkmelbourne, 2021-02-27:
@stevstrong 

I think someone else posted a similar problem in the last week, or perhaps 2 weeks ago..

Probably Apple changed something in their OS, which is no longer compatible with the hack that the bootloader uses to force re-enumeration of the USB bus.

The other possibility is defective hardware, fake chips etc etc.

I think this is something for the forum, as its not an issue with any of the core code and the core only contains a Windows driver, and this is a Mac problem, so its another reason why its not an "issue" for this repo.

If the OP doesn't reply within 24 hours I think its best to close the issue

rogerclarkmelbourne, 2021-03-06:
Closing this problem as the OP has not replied for over 1 week.

I think if people can't be bothered to reply within 1 week, we may as well close the issue, as they are obviously not actively working on their project or interested in getting a solution

closed 2021-03-06 by rogerclarkmelbourne