rogerclarkmelbourne/Arduino_STM32 #848: have an "official" install process to use with Arduino Pro IDE?

mrbbp, 2021-03-01:
In the Arduino pro IDE, the board are not listed in the board list.
with the official STM32 arduino list, the Arduino_ST7735_STM lib does not compile
Boards must be added with the board manager json file to be "installed"

have to said i like the auto completion ide...
I do not code arduino every day, so autocompletion is a good help.

i do not use the in-Arduino upload process... I prefered the STM32CubeProgrammer... have to find the compiled sketch in the temp folder (hard to find on standard MacOs interface) but the way to upload seems more easy (few option) than  in the Arduino IDE too many obscure options...

rogerclarkmelbourne, 2021-03-06:
Which core are you using.

Where did you get that library ?

stevstrong, 2021-03-07:
As explained in the forum post https://www.stm32duino.com/viewtopic.php?f=18&t=948, this core will not have Arduino board installation like method.

closed 2021-03-07 by stevstrong