rogerclarkmelbourne/Arduino_STM32 #849: Serial output garbaged

iflyhigh, 2021-03-05:
Standard example ASCIITable.ino produces garbaged serial output via USB like this:

### Example 1
ASCII Table ~ Character Map
!, dec: 33, hex: 21, oct: 41, bin: 100001
", dec: 34, hex: 22, oct: 42, bin: 100010
#, dec: 35, hex: 23, oct: 43, bin: 100011
$, dec: 36, hex: 24, oct: 44, bin: 100100
%, dec: 37, hex: 25, oct: 45, bin: 100101
&, dec: 38, hex: 26, oct: 46, bin: 100110
', dec: 39, hex: 27, oct: 47, bin: 100111
(, dec: 40, hex: 28, oct: 5101001
*, dec: 42, hex: 2A, oct: 52, bin: 101010
+, dec: 43, hex: 2B, oct: 53, bin: 101011
,, dec: 44, hex: 2C, oct: 54, bin: 101100
-, dec: 45, hex: 2D, oct: 55, bin: 101101
., dec: 46, hex: 2E, oct: 56, bin: 101110
/, dec: 47, hex: 2F, oct: 57, bin: 101111
0, dec: 48, hex: 30, oct: 60, bin: 110000
1, dec: 49, hex: 31, oct: 61, bin: 110001
2, dec: 50, hex: 32, oct: 62, bin: 110010
3, dec: 51, he110100
5, dec: 53, hex: 35, oct: 65, bin: 110101
6, dec: 54, hex: 36, oct: 66, bin: 110110
...

### Example 2

ASCII Table ~ Character Map
!, dec: 33, hex: 21, oct: 41, bin: 100001
", dec: 34, hex: 22, oct: 42, bin: 100010
#, dec: 35, hex: 23, oct: 43, bin: 100011
$, dec: 36, hex: 24, oct: 44, bin: 100100
%, dec: 37, hex: 25, oct: 45, bin: 100101
&, dec: 38, hex: 26, oct: 46, bin: 100110
', dec: 39, hex: 27, oct: 47, bin: 100111
(, dec: 40, hex: 28, oct: 50, bin: 10101010
+, dec: 43, hex: 2B, oct: 53, bin: 101011
,, dec: 44, h52, hex: 34, oct: 64, bin: 110100
5, dec: 53, hex: 35, oct: 65, bin: 110101
6, dec: 54, hex: 36, oct: 66, bin: 110110
...

Issue reproduces with some old (~5 years) BluePill board (I have 2 such boards and tested with both) with Roger's and HID bootloader or without them. Issue occurs at random places during serial output however they . At the same time core from stm32duino works as expected with this example. I've not tested with other boards since I don't have any.

rogerclarkmelbourne, 2021-03-06:
Operating system ?

stevstrong, 2021-03-06:
Sorry, but I cannot reproduce the problem with bluepill ad Arduino IDE.
Here is my output:
```
!, dec: 33, hex: 21, oct: 41, bin: 100001
", dec: 34, hex: 22, oct: 42, bin: 100010
#, dec: 35, hex: 23, oct: 43, bin: 100011
$, dec: 36, hex: 24, oct: 44, bin: 100100
%, dec: 37, hex: 25, oct: 45, bin: 100101
&, dec: 38, hex: 26, oct: 46, bin: 100110
', dec: 39, hex: 27, oct: 47, bin: 100111
(, dec: 40, hex: 28, oct: 50, bin: 101000
), dec: 41, hex: 29, oct: 51, bin: 101001
*, dec: 42, hex: 2A, oct: 52, bin: 101010
+, dec: 43, hex: 2B, oct: 53, bin: 101011
,, dec: 44, hex: 2C, oct: 54, bin: 101100
-, dec: 45, hex: 2D, oct: 55, bin: 101101
., dec: 46, hex: 2E, oct: 56, bin: 101110
/, dec: 47, hex: 2F, oct: 57, bin: 101111
0, dec: 48, hex: 30, oct: 60, bin: 110000
1, dec: 49, hex: 31, oct: 61, bin: 110001
2, dec: 50, hex: 32, oct: 62, bin: 110010
3, dec: 51, hex: 33, oct: 63, bin: 110011
4, dec: 52, hex: 34, oct: 64, bin: 110100
5, dec: 53, hex: 35, oct: 65, bin: 110101
6, dec: 54, hex: 36, oct: 66, bin: 110110
7, dec: 55, hex: 37, oct: 67, bin: 110111
8, dec: 56, hex: 38, oct: 70, bin: 111000
9, dec: 57, hex: 39, oct: 71, bin: 111001
:, dec: 58, hex: 3A, oct: 72, bin: 111010
;, dec: 59, hex: 3B, oct: 73, bin: 111011
<, dec: 60, hex: 3C, oct: 74, bin: 111100
=, dec: 61, hex: 3D, oct: 75, bin: 111101
>, dec: 62, hex: 3E, oct: 76, bin: 111110
?, dec: 63, hex: 3F, oct: 77, bin: 111111
...
```

iflyhigh, 2021-03-07:
Issue appears to be specific to my Windows 7 x64 that uses the following drivers: hhdspmc64.sys and usbser.sys. Does not reproduce with another Win 10 laptop, Linux on Raspberry or Solaris on x64 all using default USB drivers.

stevstrong, 2021-03-07:
Which IDE do you use? Which serial monitor?

iflyhigh, 2021-03-09:
> Which IDE do you use? Which serial monitor?

I'm using Arduino IDE 1.8.13 with internal serial monitor and Putty 0.69, behavior is the same in both cases.

iflyhigh, 2022-12-27:
I've just reproduced the issue on another Windows 7 machine. Tried different drivers (included into distribution and installed by Zadig) but no luck - still encounter data loss on Serial link. Latest core, Arduino 1.8.19 and some random Blue Pill. 

Issue does not reproduce with Black Pill STM32F401CCU6, but this board uses different USB driver and appears as 'STM Serial (COMxx)' in device manager while Blue Pill appears as 'Maple Serial (COMyy)'. 'Official' Arduino core also makes Blue Pill to appear as 'STM Serial' device and does not cause the observed issue. Looks like the issue is related to Maple driver and the way it supports serial communication...