rogerclarkmelbourne/Arduino_STM32 #851: Driver uninstallation

ABD-Z, 2021-03-12:
Hello,
I had installed your drivers (install_drivers.bat) and would now to uninstall them. But there is not an uninstaller. I displayed the list of installed drivers on my Windows and did not find drivers related to mapple. 
I had also used a software called installedDriversList to view drivers in a better but could not find the corresponding drivers to uninstall.
Could you please tell us how to uninstall the drivers?
Best Regards

stevstrong, 2021-03-12:
You have to open the device manager, right click on the serial (or DFU) port of the STM32 board and select "Uninstall device".

ABD-Z, 2021-03-12:
The board  should be plugged in?
In fact I want to uninstall it because i have stm32f7 which is not compatible with the drivers i installed i guess.
And I dont see serial or DFU in my device manager. 
Does that mean that the drivers were not installed correctly as I may thought?

stevstrong, 2021-03-12:
You either plug in the board and then you can identify it in device manager, or you display the hidden devices in the device manager (View->Show hidden devices) where you can check within the COM ports each the listed serial port in part Properties->Detail->Property -> Hardware ID
The value VID_1EAF means it is associated with the Libmaple driver. This you can then uninstall.

rogerclarkmelbourne, 2021-03-12:
AFIK the driver will not conflict with the STM32F7 built in DFU because they are totally different VID/PID numbers and also use totally different drivers.

The driver attache to the VID of IEAF is a CDCACM driver, and IEAF is not a VID used by anyone other than this driver.
Its not actually a registered VID. Leaflabs never registered a VID as it was too expensive to do so.

ABD-Z, 2021-03-12:
> You either plug in the board and then you can identify it in device manager, or you display the hidden devices in the device manager (View->Show hidden devices) where you can check within the COM [...]

Strangely enough, when I check the Hardware ID of my COM PORT 3, I dont have VID_1EAF... It means that when i launched the bat files it didnt install em.
Maybe because I didnt launch the bat file as admin...

rogerclarkmelbourne, 2021-03-12:
You will see the VID and PID regardless of whether the driver is installed. As otherwise the OS would not be able to query the device to get these details and then search for the correct driver.

ABD-Z, 2021-03-13:
So, If I understand the last message, you say that the OS cannot display the driver and thus I cannont uninstall it...

stevstrong, 2023-11-03:
You did not install the driver if you did not run the install batch file with admin rights.
So you probably have another driver which you can uninstall it as explained above.

closed 2023-11-03 by stevstrong