rogerclarkmelbourne/Arduino_STM32 #854: upload works from console not from ide

tomaskovacik, 2021-05-06:
hi there, 

strange issue on linux:

- clean install (arduino sw + cortex-m3 support + arduino_Stm32)

upload in ide fail, but when I remove location of arduino installation from upload command line all works fine, any hints how to fix it? I guess I am not only linux user here, so I am only one with this problem

```
nail@sid:/opt$ /home/nail/Arduino/hardware/Arduino_STM32-master/tools/linux/maple_upload ttyACM0 2 1EAF:0003 /tmp/arduino_build_28369/cdc_stm32duino_ported_k9spud_vwcdpic.ino.bin /opt/arduino-1.8.13 
Can't open /dev/ttyACM0: No such file or directory
Can't open serial /dev/ttyACM0

*** autoreset failed; you need to use RESET button
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
dfu-util: Invalid dfuse address: /opt/arduino-1.8.13

Upload failed :(
nail@sid:/opt$ /home/nail/Arduino/hardware/Arduino_STM32-master/tools/linux/maple_upload ttyACM0 2 1EAF:0003 /tmp/arduino_build_28369/cdc_stm32duino_ported_k9spud_vwcdpic.ino.bin
Can't open /dev/ttyACM0: No such file or directory
Can't open serial /dev/ttyACM0

*** autoreset failed; you need to use RESET button
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        40660 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
Waiting for /dev/ttyACM0 serial... DONE :)

```

tomaskovacik, 2021-05-07:
this works: 
```
diff --git a/STM32F1/platform.txt b/STM32F1/platform.txt
index cb5c43c..8e1a34c 100644
--- a/STM32F1/platform.txt
+++ b/STM32F1/platform.txt
@@ -118,7 +118,7 @@ tools.maple_upload.path.linux={runtime.hardware.path}/tools/linux
 tools.maple_upload.path.linux64={runtime.hardware.path}/tools/linux64    
 tools.maple_upload.upload.params.verbose=-d
 tools.maple_upload.upload.params.quiet=
-tools.maple_upload.upload.pattern="{path}/{cmd}" {serial.port.file} {upload.altID} {upload.usbID} "{build.path}/{build.project_name}.bin" "{runtime.ide.path}"
+tools.maple_upload.upload.pattern="{path}/{cmd}" {serial.port.file} {upload.altID} {upload.usbID} "{build.path}/{build.project_name}.bin"
 
 #Added tool for generic STM32 upload via serial to Serial Port 1 (pins PA9 and PA10) - note. Boot0 line needs to high on board reset to enable upload via serial
 # at the end up the upload the program is automatically run, without the board being reset
```

[image]

fpistm, 2021-05-07:
Linked to this PR
https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/814

The same PR has been submitted on the STM32 core but I do not merge it as this does not cover all cases.

PR #855 (2021-05-30): fix maple_upload dfu_address set to {runtime.ide.path} instead of {up…
…load.dfuse_addr} or nothing #854

fix #854 

tested with F1 only and only on 64bit linux, but it is simple fix
closed 2021-05-30 by stevstrong