rogerclarkmelbourne/Arduino_STM32 #856: timer configuration

mymimoms, 2021-05-27:
hello,

i want to enable real hardware PWM on pins PB8, PB9, PA15, PB3, PB10 and PB11 but anyhow none of these pins work. the chip crashes and nothing work if i use one of these pins.

it would be great to get an idea how to enable these pins. if i use the official stm32 core for the arduino environment i can use most if these pins without any problem but i need to use the maple core which i set under platformio to get the possibility using USBcomposit library.

how can i start, what should i do? thank you alot and best regards

stevstrong, 2021-05-27:
Well, for the start you should make a simple sketch to setup one of those pins as PWM and post that here.

mymimoms, 2021-05-27:
i coded complex programs with pointers, classes and structs already which work. i dont think i depents on the initial code. analogWrite(PB8, 127) do nothing, but digitalWrite(PB8,HIGH or LOW) works. i have the same issue on all other pins if i did simple analogWrite example for each pin seperately

mymimoms, 2021-05-27:
i already asked the pio community and they lead me to this file
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/bab4fda0fa31ba2d1f91f45329843d2df77da74b/STM32F1/variants/generic_stm32f103c/board.cpp#L75-L90

it seems i need to change some stuff here. i tried to add &Timer4 to PB10 and PB11 .. but it dont work either

stevstrong, 2021-05-28:
Again, make a simple sketch to setup one of those pins as PWM, test it, and if it does not work then post it here.
Otherwise I cannot image what you did and what eventually may be wrong.
Also, the best place to discuss about such things is the forum: www.stm32duino.com

mymimoms, 2021-07-08:
byte cnt;

void setup()
{
 pinMode(PA15, PWM);
 pinMode(PB3, PWM);
 pinMode(PB10, PWM);
 pinMode(PB11, PWM);
}

void loop()
{
    analogWrite(PA15, cnt);
    analogWrite(PB3,  cnt);
    analogWrite(PB10, cnt);
    analogWrite(PB11, cnt);
   cnt++;
}

mymimoms, 2021-07-08:
got it:

boards.cpp

extern const stm32_pin_info PIN_MAP[BOARD_NR_GPIO_PINS] = {

    {&gpioa, &timer2, &adc1,  0, 1,    0}, /* PA0 */
    {&gpioa, &timer2, &adc1,  1, 2,    1}, /* PA1 */
    {&gpioa, &timer2, &adc1,  2, 3,    2}, /* PA2 */	
    {&gpioa, &timer2, &adc1,  3, 4,    3}, /* PA3 */
    {&gpioa,   NULL, &adc1,  4, 0,    4}, /* PA4 */
    {&gpioa,   NULL, &adc1,  5, 0,    5}, /* PA5 */
    {&gpioa, &timer3, &adc1,  6, 1,    6}, /* PA6 */
    {&gpioa, &timer3, &adc1,  7, 2,    7}, /* PA7 */
    {&gpioa, &timer1, NULL,  8, 1, ADCx}, /* PA8 */	
    {&gpioa, &timer1, NULL,  9, 2, ADCx}, /* PA9 */	
    {&gpioa, &timer1, NULL, 10, 3, ADCx}, /* PA10 */
    {&gpioa, &timer1, NULL, 11, 4, ADCx}, /* PA11 */
    {&gpioa,   NULL, NULL, 12, 0, ADCx}, /* PA12 */	
    {&gpioa,   NULL, NULL, 13, 0, ADCx}, /* PA13 */	
    {&gpioa,   NULL, NULL, 14, 0, ADCx}, /* PA14 */
    {&gpioa,   &timer2, NULL, 15, 1, ADCx}, /* PA15 */	

    {&gpiob, &timer3, &adc1,  0, 3,    8}, /* PB0 */	
    {&gpiob, &timer3, &adc1,  1, 4,    9}, /* PB1 */
    {&gpiob,   NULL, NULL,  2, 0, ADCx}, /* PB2 */
    {&gpiob, &timer2, NULL,  3, 2, ADCx}, /* PB3 */
    {&gpiob,   NULL, NULL,  4, 0, ADCx}, /* PB4 */
    {&gpiob,   NULL, NULL,  5, 0, ADCx}, /* PB5 */
    {&gpiob, &timer4, NULL,  6, 1, ADCx}, /* PB6 */
    {&gpiob, &timer4, NULL,  7, 2, ADCx}, /* PB7 */	
    {&gpiob, &timer4, NULL,  8, 3, ADCx}, /* PB8 */	
    {&gpiob, &timer4, NULL,  9, 4, ADCx}, /* PB9 */	
    {&gpiob, &timer2, NULL, 10, 3, ADCx}, /* PB10 */	
    {&gpiob, &timer2, NULL, 11, 4, ADCx}, /* PB11 */
    {&gpiob,   NULL, NULL, 12, 0, ADCx}, /* PB12 */
    {&gpiob,   NULL, NULL, 13, 0, ADCx}, /* PB13 */
    {&gpiob,   NULL, NULL, 14, 0, ADCx}, /* PB14 */
    {&gpiob,   NULL, NULL, 15, 0, ADCx}, /* PB15 */
    {&gpioc,   NULL, NULL, 13, 0, ADCx}, /* PC13 */	
    {&gpioc,   NULL, NULL, 14, 0, ADCx}, /* PC14 */
    {&gpioc,   NULL, NULL, 15, 0, ADCx}, /* PC15 */
};

extern const uint8 boardPWMPins[BOARD_NR_PWM_PINS] __FLASH__ = {
    PB0, PB1, PA7, PA6, PA3, PA2, PA1, PA0, PB7, PB6, PA11, PA10, PA9, PA8, PA15, PB3, PB10, PB11, PB8, PB9
    
  boards.h
  #define BOARD_NR_PWM_PINS         20
  
  in setup:
  afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY);
  afio_remap(AFIO_REMAP_TIM2_FULL);

adi-candra, 2021-08-04:
hi, how to turn on timer1 complementary ?

stevstrong, 2023-11-03:
Closing this, as it seems to be resolved.

closed 2023-11-03 by stevstrong