rogerclarkmelbourne/Arduino_STM32 #857: 128 Kb CPU hangs up when 128Mhz overclocking is applied STM32103

pirvcb, 2021-06-08:
In arduino 1.8.3, when i select the 128 Mhz overclocking option CPU hangs up for 128 Kb memory version; the same code runs ok at 72 legacy frequency. In 64 Kb version, at 128 Mhz, everything is ok. 
Both chips are original STM and the Arduino compiler gives back NO ERRORS for both cases.
I cant understand why...

rogerclarkmelbourne, 2021-06-08:
Operation when overclocked is not guaranteed.

You won't get compile errors, as the clock speed is a runtime setting.

It's also possible that although you think the chips are genuine STM they may not be. It's impossible to see the differences between fake and real STM32 chips without decapping the chip and using a high power microscope

pirvcb, 2021-06-09:
Thank you so much, Roger, for your clean reply.
Just one last question:
Could you point me, in Arduino/hardware/STM32 etc installation, WICH FILE(S) contains the PLL multiplier factor to let me try to get intermediate speed velocities?
I mean, now i can select only 72 or 128 Mhz. 
Because i just need get out a little bit more power from the platform for running a Mozzi sketch ( https://sensorium.github.io/Mozzi/doc/html/ ), i would like try to slow down a little from 128 to see if this gets the micro running.
Thnks again for your time.
Davide

rogerclarkmelbourne, 2021-06-09:
You can set other PLL multipliers

The clock crystal is 8Mz, so you can configure for 72,80,88,96,104,112,120 etc

See 

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/bab4fda0fa31ba2d1f91f45329843d2df77da74b/STM32F1/variants/generic_stm32f103c/wirish/boards_setup.cpp#L50-L60

closed 2021-06-09 by rogerclarkmelbourne