rogerclarkmelbourne/Arduino_STM32 #858: I2C2 Issue - same response from sensor

shayne243, 2021-06-21:
I'm using an Atlas RGB sensor connected to an STM32 board running maple 14.0.0 using I2C2. After issuing a command to the sensor using write(), I receive the expected response. However, with every successive command that I issue, I receive the same initial response despite changing the command.

For instance, if I send a command to return the device information, it does that successfully. Then, if I send a command to take a reading, it responds with the device information instead of the reading. This happens for any command I send after that.

I am sure it is not a problem with the sensor itself as I have tested it using the STMArduino core instead of maple and it worked fine as well as on an Arduino UNO.

The issue seems to be in the implementation of Wire.h / WireBase.h.

dewhisna, 2021-06-21:
Try switching to `WireSlave.h` to pull in the alternate library and see if it makes a difference.  `WireSlave` was designed as a replacement for `Wire`, only able to do both master and slave operations instead of master only.  If it works, then the issue may be in the `Wire` library itself rather than the underlying low-level code.

shayne243, 2021-06-21:
Yes that worked

stevstrong, 2023-11-03:
Seems to be resolved, so I will close this.

closed 2023-11-03 by stevstrong