rogerclarkmelbourne/Arduino_STM32 #865: How to write code for and program cz ministm32f103zx board ?

eagl1, 2022-01-16:
Hi,

I have this board which is cz ministm32f103zx development board:

[https://sparklogic.ru/wp-content/uploads/2019/11/img_5de0f6ad10544.jpg](url)

How to program it ?

eagl1, 2022-01-19:
I solved the usb port issue:

[image]

But in Arduino IDE, I get this error:

> Arduino: 1.8.15 (Windows 10), Board: "Generic STM32F103Z series, STM32F103ZE, Serial, 72Mhz (Normal), Smallest (default)"
exec: "/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Generic STM32F103Z series.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

eagl1, 2022-01-20:
OK, I was able to program it with Arduino IDE.

First solve the "pl2303hxa phased out since 2012 please contact your supplier" issue, with updating the driver with this file:

http://www.mediafire.com/file/982x6iyk89v95dp/Prolific_PL2303_driver_v3.3.2.102_%25282008-24-09%2529_Win8_x64_x86.7z/file

With the following setting.

During flashing the MCU with Flash Loader Demonstrator with the appropriate bin file:
1. Set boot1 to GND.
2. Power the board and select Next, then the next screen show the size of the flash memory.
3. Select Next, this page has the flash mapping. Select Next.
4. In the final page, select Download to device and check "Verify after download" and mount the bin file and select Next and wait until flashing process is done.

In Arduino IDE:
1. Choose the board type as "Generic STM32F103Z series"
2. In variant option select "STM32F103ZE"
3. Upload method, select "Serial"
4. CPU speed ""72MHz"
5. Optimized "Smallest"
6. Select the required COM port
7. Remove the jumper from boot1 pin
8. Reset the board
9. Then program blink example

Still have an issue then I couldn't use the LED on PE1 pin.

rogerclarkmelbourne, 2022-09-17:
Close as solved

closed 2022-09-17 by rogerclarkmelbourne

eagl1, 2022-09-17:
I'm really happy I was able to communicate with the board.

I also bought the ST LINK V2, now I can program it with Arduino and the other programming platforms that support this programmer.

rogerclarkmelbourne, 2022-09-17:
ST Link is definitely worth buying.

You can also use it with the STM32 Cube so that the ST Link allows in circuit debugging

eagl1, 2022-09-17:
Yep, thanks man for the support. I really like your work in the Arduino_STM32.

I have a question about programming STM32 boards on Arduino vs on the more professional platform; like, STM32 Cube. Is there a difference in performance ?