rogerclarkmelbourne/Arduino_STM32 #866: Wiki page for Installation has a broken link

alanbjohnston, 2022-01-23:
Hi,  the wiki page for Installation has a broken link. 

The page is https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation 

The text with the broken link is
Download zip file containing the STM32 files from here.

Thanks for the great software!

wraybowling, 2022-01-24:
Came here to report the same problem

fpistm, 2022-01-24:
Seems the wiki has been modified wile it should not.
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation/_compare/ec0d552ae0f8a2cb16fa9158cb10b81ce211ef23...6b2e8117fcc3dfd36fb106f761995977d510e808

Old link should be correct and should be restored.

wraybowling, 2022-01-24:
I fixed it. I verified that it's simply a zip of this repository. The wiki was using a URL shortener

closed 2022-09-16 by rogerclarkmelbourne