rogerclarkmelbourne/Arduino_STM32 #871: Wire.h compile warnings started appearing for I2C

i998, 2022-02-23:
I updated my copy of Arduino_STM32 to the latest version from the repository and I started getting the following compile warnings  on the code that uses I2C   - including   I2C examples, my old code that compiled normally before etc. Target board is Maple Mini. Arduino IDE 1.8.19. OS is Win7.

On restoring  my old copy of Arduino_STM32 from approx Feb2021 to C:\Users\UserA\Documents\Arduino\hardware   these errors stoped appearing.

Here  is the output for example i2c_scanner_wire from the repository examples:

```

Compiling sketch...
"C:\\Users\\UserA\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_maple_mini -DVECT_TAB_ADDR=0x8005000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10819 -DARDUINO_MAPLE_MINI -DARDUINO_ARCH_STM32F1 -DMCU_STM32F103CB -DSERIAL_USB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -DSERIAL_USB -mthumb -march=armv7-m -D__STM32F1__ "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple" "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/include" "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/stm32f1/include" "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/stm32f1" "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple/usb/usb_lib" "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\cores\\maple" "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\variants\\maple_mini" "-IC:\\Users\\UserA\\Documents\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\libraries\\Wire" "C:\\Users\\UserA\\AppData\\Local\\Temp\\arduino_build_820807\\sketch\\i2c_scanner_wire.ino.cpp" -o "C:\\Users\\UserA\\AppData\\Local\\Temp\\arduino_build_820807\\sketch\\i2c_scanner_wire.ino.cpp.o"

In file included from C:\Users\UserA\Documents\Arduino\hardware\Arduino_STM32\STM32F1\libraries\Wire/Wire.h:42:0,
                 from C:\Users\UserA\Documents\Arduino\hardware\Arduino_STM32\STM32F1\libraries\Wire\examples\i2c_scanner_wire\i2c_scanner_wire.ino:27:

C:\Users\UserA\Documents\Arduino\hardware\Arduino_STM32\STM32F1\libraries\Wire/utility/WireBase.h: In member function 'size_t WireBase::write(const uint8*, int)':

C:\Users\UserA\Documents\Arduino\hardware\Arduino_STM32\STM32F1\libraries\Wire/utility/WireBase.h:124:69: warning: no return statement in function returning non-void [-Wreturn-type]
     size_t write(const uint8* buf, int nr) { write((uint8*)buf,nr); }
                                                                                             ^

```
Could someone have a look please?

GitMoDu, 2022-02-24:
```
warning: no return statement
     size_t write(const uint8* buf, int nr) { write((uint8*)buf,nr);
```
The compiler's not wrong.

Seems like this was added recently in https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/098a98c30e01eeb3f1cf9e9589a74a9263251e7c

stevstrong, 2022-02-24:
Shold be solved by https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/0ddc946f6fbda1968de9f4f0aec2bf527a08fd0a

closed 2022-02-24 by stevstrong