rogerclarkmelbourne/Arduino_STM32 #876: USB HID Joystick together with CompositeSerial

Joshy-2010, 2022-04-03:
Hello,
I am trying to use the USBComposite Library built into this STM32 core.

I have succesfully programmed myself a HID joystick ontop of the "simplejoystick" example sketch:
` USBHID HID;
  HIDJoystick Joystick(HID);

  void setup() {
   HID.begin(HID_JOYSTICK);
   while (!USBComposite);
  }
`
......

AndI also got the multiserial eample to work, both seperately. The next step is to combine this two into a single programm: A joystick to which I can write serial commands to activate some LEDs and other small gimics. I saw the "keyboardwithleds" example sketch (which also worked perfectly) and thought I could just replace "HID_KEYBOARD" with "HID_JOYSTICK":

` USBHID HID;
  HIDJoystick Joystick(HID);
  USBCompositeSerial USB_Serial;

  void setup() {
  HID.begin(CompositeSerial, HID_JOYSTICK);
   while (!USBComposite);
  }
`
......

Unfortunately, my Pcsays that it could not recognize the usb device anymore. Is joystick + serial not supported? Is there another way to send joystick data both from the STM32 to the Pc while also sending small commands back to the STM32?
Do I have to call a few other commands to start everything up? I saw a few other functions like "plugin2.registerComponent();" but I dont really understand that they are doing differently.

Thanks in advance

Joshy-2010, 2022-04-09:
I think I have figured it out. These are the predeclared variables:

`const uint8_t reportDescription[] = {
   HID_JOYSTICK_REPORT_DESCRIPTOR(),
};

USBCompositeSerial USB_Serial;
USBHID HID;
HIDJoystick Joystick(HID);`

And this needs to go into "void setup":
`HID.begin(USB_Serial, reportDescription, sizeof(reportDescription));
  while (!USBComposite);`

Strangly, it still reports as a keyboard, mouse and game controller in windows, even though I have only specified game controller in the "reportDescription".

rogerclarkmelbourne, 2022-09-16:
Thanks. Closing as resolved

closed 2022-09-16 by rogerclarkmelbourne