rogerclarkmelbourne/Arduino_STM32 #877: Add serial upload support for STM32F4 boards

XanCraft21, 2022-04-07:
I have an STM32F1 board that has a on-board USB to UART chip, which makes programming a breeze using this arduino core. I thought i would buy an STM32F4 board since this library supports it, but there isn’t any serial upload support. I thought that the boot loader would work, but my device does not have a valid port. I am not able to reprogram the board, even after trying many fixes for corrupted drivers. I was wondering if you could add serial upload support to the STM32F4 boards to make it easier to program, since my boot loader is not correctly initialized. Thank you to anyone who can help.

board707, 2022-04-08:
There is no Serial upload for STM32F4, but you can setup DFU upload with same dfu-util as for STM32F1 via onboard USB. But using st-link is more easy and convenient, as i think

XanCraft21, 2022-04-08:
I have some questions relating to that, how do i add a software port with DFU to a board that doesn’t come with one (it has the DFU software already on it but no port to select), how do i fix a driver that isn’t made correctly and doesn’t correctly install (which might have no affect on the situation), does this software automatically detect if an St-link is plugged in during uploading, and do i need drivers for that St-link while using this arduino core (does this come with it’s own St-link driver)? I do not feel safe downloading the official drivers or anything from STM, so i hope i’m already prepared as soon as i downloaded this core.

board707, 2022-04-08:
Do not confuse upload via St-link and DFU, they are two different methods. St-link upload is enabled by default for STM32F4, st-link drivers are included in repo (see the tools/win/stlink/ folder).

XanCraft21, 2022-04-09:
Ok, thank you. I will try the st-link. I currently do not own one as I thought I didn’t need one, and i’m still looking for a good place to buy a generic version online. I still hope that one day serial support will come out for the F4 boards.

XanCraft21, 2022-08-27:
Sorry it took me forever to finish this. I’ve decided to go a different way of programming stm32f4 boards. I got a generic serial FTDI TTL converter and modified the boards and platform txt files to add the stm32f1 serial upload protocol and it worked perfectly. Using pins PA9 and PA10 to program (TX and RX flipped from serial chip) and then i can print to the Serial monitor using Serial1. I will prepare to close this issue here soon.

closed 2022-08-29 by XanCraft21