rogerclarkmelbourne/Arduino_STM32 #879: There's no Maple Serial but only Maple DFU in the Device Manager

W-William, 2022-04-25:
Hi, I've got a question to ask
I wanna use my bluepill( STM32F103C6T6) via Arduino IDE with a micro-USB cable only
So I followed guidance on google, installed the 2 drivers on my Win11 and flashed the bootloader "_generic_boot20_pc13.bin_" to my bluepill, 
but in the Device Manager, there isn't any new device named "_Maple Serial_" but only one named "_Maple DFU_" 
[image]
and I still can't use my bluepill via Arduino IDE cuz there's no Serial COM port available 
what should I do? plz help me. thx.

JiriBilek, 2022-04-25:
I haven't used this core and bootloader for a long time, but AFAIK you don't need a serial port for DFU flashing. Try to select whatever COM port you have.

rogerclarkmelbourne, 2022-09-16:
If the board does not have the bootloader installed it won't appear as the Maple DFU device

closed 2022-09-16 by rogerclarkmelbourne

llmpowerll, 2022-11-08:
Hi rogerclarkmelbourne
 I have this problem too. (my windows 10 ,  Arduino IDE 2.0.1 , Upload Methode = STM32duino Bootloader)
I burn BootLoader to STM32 by Flash Loader and install two .bat file driver and device manager shows DFU Serial (COMx),
but when I try Program STM32 with Arduino IDE every things ok before final reset during programming process.
I mean when programing process finish and final software reset step, the DFU Serial (COMx) changes to Maple DFU and serial port ???!!!
Serial Port disappear and never works again.(I checked this problem with hardware reset button and pin and every things was OK and never destroy Serial Port)
I believe that Software resetting at the final step of programing make bad effect on Bootloader segment and destroy it.

Please Help me to solve this problem.
best regard.

board707, 2022-11-08:
Hi @llmpowerll 
This is not a software problem, but a hardware one. On bluepill clones, the USB is incorrectly wired, so it has problems with interacting with the bus.

Try this:
Start uploading code in Arduino IDE WITHOUT connect the board. When the compilation is finished and in log window appears message like "Looking for DFU device...." - only in that time connect the bluepill to the USB port. You have about 5 seconds to do this.

I helps in my case and DFU upload always goes fine.

llmpowerll, 2022-11-08:
thanks to your fast and clear replay.
my BluePill device is original and its not clone or fake MCU or PCB.  
I was able to program Bluepill in Arduino IDE. it had two important reasons. 
1- install .bat file in "Driver Signature Enforcement" mode. and I could Program bluepill 4 time and after that again COM port disappear so again burnt Bootloader File and  test it. the result repeated for  four time programing.
2- I install Java Runtime Environment jre8-u241 and test it. Everything's OK and I can Program it with out any problem (test it 10 time on blink Example and after restart system and boot normally everything's works OK and the problem solved )
Pay attention I test latest version of  Java Runtime Environment jre8-u341 and it didn't create COM port for Programing.
Also test Java Runtime Environment jre8-u151 (older version) and it work only for four time programing and after that COM port disappear.

Conclusion : 
1- try install drivers in  "Driver Signature Enforcement" mode
2- check your installed Java Runtime Environment the best version is Java Runtime Environment jre8-u241 which I tested and works correctly.

best regard