rogerclarkmelbourne/Arduino_STM32 #880: i2c freeze MCU

ciko2k, 2022-05-11:
Hi, i'm using the wire scan example. if my i2s peripheric is connect the code run well, but asap i disconnect it, MCU freeze.
could someone help me?

`
#include <Wire.h>
#include <libmaple/iwdg.h>

//use IIC2

TwoWire WIRE2 (2,I2C_FAST_MODE);
#define Wire WIRE2

#define TXPIN   PA8

int cycle;

void setup() {
 iwdg_init(IWDG_PRE_32, 8000);
 iwdg_feed();

  pinMode(TXPIN, OUTPUT);
  digitalWrite(TXPIN,HIGH);

  Serial1.begin(9600);
  Wire.begin();
  delay(10);
  Serial1.println("*****");
  Serial1.println("\nI2C Scanner");
  Serial1.println("****");
  delay(10);
}

void loop() {
  iwdg_feed();
  byte error, address;
  int nDevices;

  Serial1.print(cycle);
  Serial1.println(". Scan ***** ");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    //Serial1.println(address); delay(1);
    
    if (error == 0) {
      
      Serial1.print("I2C device found at address 0x");
      if (address < 16) 
        Serial1.print("0");
      Serial1.println(address);

      nDevices++;
    }
    
    else if (error == 4) {
      Serial1.print("Unknown error at address 0x");
      if (address < 16) 
        Serial1.print("0");
      Serial1.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    Serial1.println("No I2C devices found");
  else
    Serial1.println("done");

  delay(1000);           // wait 5 seconds for next scan
  cycle++;
  
}
'

GitMoDu, 2022-05-11:
Without external pull-ups (like the ones likely in your device) I2C will stall because the lines won't raise up.

rogerclarkmelbourne, 2022-09-16:
Closed due to inactivity from OP

closed 2022-09-16 by rogerclarkmelbourne