rogerclarkmelbourne/Arduino_STM32 #881: USBSerial does not start

ZavenArra, 2022-07-07:
I am attempting to use the USBSerial object provided in the USBComposite library.  However, USBSerial never starts.   I have ground, D+ and D- (PA12 and PA11 on the F103RB Nucleo) from the STM32F103RB pinout connected to a USB cable that is plugged into my computer.   The gist below shows how I am using the code, which I believe is correct - I tried both ways recommended in the README.  I also verified that USBCLK is running at 48MHz.  The code simply sits in the while loop.   I also dug into the library at little bit, and I can see that the USB_CONNECTED and USB_CONFIGURED booleans never change.  What is missing here?  Is this a problem with my code, or an issue with the library support?

```

  USBComposite.clear(); // clear any plugins previously registered
  USBSerial.registerComponent(); 
  USBComposite.begin();
  while(!USBComposite)
  {
    Serial2.println("waiting");
    delay(1000);
  }
  
  
  or 
  
  USBSerial.begin();
  while(!USBSerial)
  {
    Serial2.println("waiting");
    delay(1000);
  }
```

https://gist.github.com/ZavenArra/bd261c8eb6cbd8e77ec94c39e0f29eb7

JiriBilek, 2022-07-07:
In all schematics around F103 there is a resistor from D+ to +5V. Do you have it there? I guess it is needed for USB Enumeration process.

ZavenArra, 2022-07-12:
Thanks, that was it!

closed 2022-07-12 by ZavenArra

ZavenArra, 2022-07-12:
@JiriBilek Where did you review schematics to come to this conclusion?

JiriBilek, 2022-07-12:
You can find it easily, just google "bluepill schematics". One for all: https://microcontrollerslab.com/stm32f103c8t6-blue-pill-pinout-peripherals-programming-features/

ZavenArra, 2022-07-13:
Oh I have the nucleo schematics, I meant how you knew to add the resistor.  The blue pill schematic is actually what I did want, didn't think of that.  Good reference for building a system off F103.  Thanks!

JiriBilek, 2022-07-13:
I checked the pullup resistor value - it should be **1k5** according to e.g. https://forum.arduino.cc/t/blue-pill-usb-and-1-8k-resistor/606316
I have several boards with 10k resistors and they generally work but sometimes, the enumeration fails. So please use 1k5.

JiriBilek, 2022-07-15:
One more follow-up: the resistor should be to 3.3V - see 
https://community.st.com/s/question/0D50X00009XkY1uSAF/stm32f103-usb-circuit
But for me 5V was always fine.