rogerclarkmelbourne/Arduino_STM32 #882: mString no support from stm32

zDimaBY, 2022-08-04:
https://github.com/GyverLibs/mString/issues/3 I don't quite understand what prevents you from using this library.
Perhaps you need to include some other libraries?

board707, 2022-08-04:
@zDimaBY 
What does the mString library have to do with this project? I think it's the author of that library who should care why his library doesn't work with Arduino_STM32

zDimaBY, 2022-08-04:
> Perhaps you wanted to write "me" instead of "you"?

Yes, But on arduino compiler it works, for example simple gcc function: itoa (int data, char* string, int radix), ltoa (long data, char* string, int radix), ultoa (unsigned long data, char* string, int radix) the compiler says: 'utoa' was not declared in this scope. When you change the target to an Arduino board, everything compiles. That's why I asked the question..

board707, 2022-08-04:
itoa is a non-standard function which is supported by some compilers. Going by the error, it seems that not supported by Arduino_STM32. 
But i will check it later and post back

board707, 2022-08-04:
I checked - itoa() function is supported. No additional header files are required to be included. 
The sketch below compiles fine
```
void setup() {
 char a[22];
 int b=33;
 itoa(b,a,10);
}

void loop() {}
```

I think you need to look for errors in your code.

zDimaBY, 2022-08-04:
[image]

What version of IDE are you using?

board707, 2022-08-04:
I tested it in v.1.8.3
For Arduino v.1.8.13 and above try to add at beginning of file

#include "itoa.h"

rogerclarkmelbourne, 2022-09-16:
Closed due to inactivity from OP

closed 2022-09-16 by rogerclarkmelbourne