rogerclarkmelbourne/Arduino_STM32 #883: random() without parameter like in standard Arduino API is missing.

ArminJo, 2022-08-07:
Standard Arduino programs do not compile if they use` random();`

E.g.
```
/home/runner/Arduino/libraries/NeoPatterns/src/NeoPatterns.hpp:1474:31: error: no matching function for call to 'random()'
         tRandom.Long = random();
                               ^
/home/runner/Arduino/libraries/NeoPatterns/src/NeoPatterns.hpp:1474:31: note: candidates are:
...
/home/runner/.arduino15/packages/stm32duino/hardware/STM32F1/2021.5.31/cores/maple/wirish_math.h:50:6: note: long int random(long int)
 long random(long max);
```

if I replace the line 50 in wirish_math.h
`long random(long max);`
with
`long random(long max = __LONG_MAX__);`
everything works fine.

Do you require a PR?

rogerclarkmelbourne, 2022-09-16:
Where is random() implemented ?

ArminJo, 2022-09-17:
??? 
long random(long max = __LONG_MAX__); introduces a default/optional parameter. 

long random(long howbig) is implemented [here](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0ddc946f6fbda1968de9f4f0aec2bf527a08fd0a/STM32F1/cores/maple/wirish_math.cpp#L33)

rogerclarkmelbourne, 2022-09-17:
Create a PR and I will merge it, as I think its a low risk change

ArminJo, 2022-09-17:
Done!

ArminJo, 2022-09-17:
Thank YOU for maintaining this core 🥇 .

closed 2022-09-26 by rogerclarkmelbourne

stevstrong, 2024-03-15:
which PR?

ArminJo, 2024-03-15:
Sorry I had "Tomaten auf den Augen". 
it was PR https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/889.
I was searching for it in order to delete my local fork and I have not found it, since I looked at the wrong place.
Thanks again for maintaining!

commit 2f43c05e (2022-09-26): fixes #883 

 random() without parameter like in standard Arduino API is missing.