rogerclarkmelbourne/Arduino_STM32 #884: Why is itoa.h missing in the includes and thus disabling use of utoa() function?

ArminJo, 2022-08-07:
I am using FQBN: stm32duino:STM32F1:genericSTM32F103C

**Using utoa(aNumber, tStringBuffer, 10); throws error: 'utoa' was not declared in this scope**
When I add a line `#include "itoa.h"` in wirish.h
```
#include <avr/interrupt.h>
#include "itoa.h"

#include <libmaple/stm32.h>
```
the error vanishes.

Do you require a PR?

board707, 2022-08-07:
Because itoa() and utoa() are not the part of C++ standard library and it support are not obligatory.
You should be grateful for that itoa.h is available :)

rogerclarkmelbourne, 2022-09-16:
This is not a support forum.

Try www.stm32duino.com

ArminJo, 2022-09-17:
This is not a support question.

stevstrong, 2023-11-03:
If you have a solution then please post a pull request.

board707, 2023-11-03:
The author writes that ones need to add one line

stevstrong, 2023-11-03:
If the author has a problem and a solution, then it should also take time to post a PR.

closed 2023-11-03 by ArminJo