rogerclarkmelbourne/Arduino_STM32 #885: Missing wiring_private.h file for stm32f4 boards

XanCraft21, 2022-08-29:
Hello, i just discovered that the file wiring_private.h is missing in the stm32f4 boards. I don’t know if anyone else has opened an issue for this yet. Some libraries need this file to work like the display drivers by Adafruit. If anyone can add this file i will gladly appreciate it. Thank you to anyone who can help.

stevstrong, 2022-08-29:
Do you have an example?
Try to comment out this header file, it should not be needed for F4 boards.
Or you can simply create a dummy file.

XanCraft21, 2022-08-29:
Commenting out the file or adding a dummy file fixes that error, but there are a bunch of other ones. It seems like the library Adafruit_GFX does not like the I2C drivers in the stm32f4 library. I might change the name of this issue for this. I hope this gets fixed soon.

board707, 2022-08-30:
Try to use Adafruit GFX library version prior to 1.8.0, for example [ Adafruit GFX 1.7.0](https://github.com/adafruit/Adafruit-GFX-Library/releases/tag/1.7.0)

rogerclarkmelbourne, 2022-09-16:
Closed due to inactivity from OP

closed 2022-09-16 by rogerclarkmelbourne

XanCraft21, 2022-09-17:
Sorry i have forgotten to get back to this. The older version of the library works so so. Maybe it’s ok for now. Thank you for the help.

rogerclarkmelbourne, 2022-09-17:
OK