rogerclarkmelbourne/Arduino_STM32 #886: Suspicious link in the Wiki -> Installation section

Stanimir-Petev, 2022-09-02:
Hello!

I tried to install the package following the instructions shown here: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

And when I get to download and extract the archive I noticed that it took waaaaay too long to extract. When I checked the folder it turns out it was like 16 GB (GIGA bytes)! As I checked the content of the folder I noticed a number of *.exe files nested inside of it and the folders down the tree. This didn't really make sense to me having them there but initially I thought maybe they are needed (the names seemed legit though). As I asked a colleague who tested the same package a week or so ago his folder was ~70 MB.

Then when I checked the link by hovering over the link and tried to copy it it was this one (not sure if we should even open it): https://github.com/lampii/VideoCaptureUtility/releases/download/42/Arduino_STM32-master.zip
The part with "VideoCaptureUtility" immediately took my attention and I tried to just check the repository itself but strangely it doesn't exist. When you tried to open: "https://github.com/lampii/VideoCaptureUtility" it actually changes into https://github.com/olyafro/VideoCaptureUtility

Then I checked the revisions of that page and it seems that happened 3 days ago. Before that the link lead to: https://github.com/rogerclarkmelbourne/Arduino_STM32/archive/master.zip which I assume is the correct link but after that commit it is the link shown earlier.

I HIGHLY doubt this is intentional and is supposed to be like that (16 GB for a package seems extremely high). Is it possible for someone with permissions to investigate what's the deal with this link and how it get there?

rogerclarkmelbourne, 2022-09-02:
This does looks suspicious

rogerclarkmelbourne, 2022-09-02:
It looks like various bad actors have been making changes to the Wiki

I did not realise that the Wiki could be changed by anyone other than the admins :-(

I'll have to attempt to lock the wiki from being changed by anyone other than the admins, and revert the various changes by the bad actors.

I'm not sure how to report the bad actors to github, but I doubt its possible to trace who they really are

rogerclarkmelbourne, 2022-09-02:
OK. I now changed the default setting, to not allow any github user to change the wiki. IMO this is big mistake by github to allow anyone who registers an account to endit any wiki, and github does not even send out any notification to the account owner to notify them of changes to the wiki

board707, 2022-09-02:
@Stanimir-Petev 
Do you test this large archive for the viruses?
@rogerclarkmelbourne 
As I see the link was changed by [WhyNut user ](https://github.com/whynut)
You can report your behavior to GitHub admins by special link at the left on his GitHub personal page (something like "Block or Report" link)

Stanimir-Petev, 2022-09-02:
@rogerclarkmelbourne  thanks for the quick response!
Hopefully it did no harm to anyone. Although it could have been done in other sections of the wiki so it's worth checking there too. It's just that I was looking for these instructions. To be fair the one who made that went way too far and made it quite obvious. I mean come on from a 70MB to 16 GB.... not sure how did he expected to go unnoticed. Maybe he should have been more subtle :D
Anyway I am glad that it was sorted out and yes it's strange that by default everyone can edit the text and as you mentioned without even sending any kind of notifications.

@board707 I didn't. After I checked out what's inside the folder I deleted it right away. Can't claim that there were viruses but I wouldn't be surprised at all if there were.

rogerclarkmelbourne, 2022-09-02:
@ board707 

Thanks. I've reported that user to github

I'm now reviewing all the pages in the wiki, but so far I did not see any other problems

Stanimir-Petev, 2022-09-02:
Great!
Well with this out of the way I guess this issue could be closed :)

rogerclarkmelbourne, 2022-09-02:
I'll leave it open for a while, becuase I've referenced this issue in my report to github

It looks like in the past other bad actors have also changed the installation page in the past.

rogerclarkmelbourne, 2022-09-02:
I've manually checked the history of each page, and I think only the installation page was changed.

This has now been fixed.

Stanimir-Petev, 2022-09-02:
OK, thanks and have a nice day :)

rogerclarkmelbourne, 2022-09-02:
@Stanimir-Petev 

No worries. 

Thanks very much for reporting the problem.

board707, 2022-09-02:
@rogerclarkmelbourne 

I would like to take this opportunity to ask you not to abandon this project. This is a great job and in many ways it still better than the official core from STM. At least your code is more clear to me :)

Even if there is no further development, the support of the author still gives a lot.
Thanks again

rogerclarkmelbourne, 2022-09-02:
@board707 

I don't have much time to devote to this, as I have other projects I'm working on

@stevstrong  has been continuing my work, but he also probably has many other things in his life

board707, 2022-09-02:
So I understand.
Thank you

stevstrong, 2023-11-03:
I will try to give support as long as my time allows.

closed 2023-11-03 by stevstrong