rogerclarkmelbourne/Arduino_STM32 #902: ADC DMA Not work properly (stm32f1)

satar1980, 2023-03-08:
i want to take 1500 square wave data at pin A0 and 1500 sine wave data at pin A1, but  the output is always swapped. so it produce 2 graphs that merge each other. what should be graph 1 is a square wave and graph 2 is a sine wave. is there anyone who can fix it.

//==============================================================

#include <STM32ADC.h>
uint8 pins[] = {PA0, PA1};
uint8 pin = PA0;

uint16_t dataPoints[3000];

char dataready = 0;

#define sampleFreqKhz       1
#define samplePeriodus      1600 / sampleFreqKhz

volatile static bool dma1_ch1_Active; //flag for interrupt

STM32ADC myADC1(ADC1);

void tx (unsigned char c) {
  //Serial.print(c);
  Serial.write( byte(c));

}

int ledState = LOW;
volatile unsigned long blinkCount = 0;

void blinkLED(void)
{
  if (ledState == LOW) {
    ledState = HIGH;
    blinkCount = blinkCount + 1;
  } else {
    ledState = LOW;
  }
  digitalWrite(PB1, ledState);
}

void setup() {
  pinMode(PC13, OUTPUT);
  digitalWrite(PC13, HIGH);
  Serial.begin(115200);
  pinMode(PA0, INPUT_ANALOG);
  pinMode(PA1, INPUT_ANALOG);
  pinMode(PB1, OUTPUT);

  Timer3.pause();
  Timer3.setPrescaleFactor(1);
  Timer3.setOverflow(4);
  Timer3.setMasterModeTrGo(TIMER_CR2_MMS_UPDATE);
  Timer3.refresh(); // обнулить таймер
  Timer3.resume(); // запускаем таймер

  Timer1.init();
  Timer1.pause();
  //Timer1.setMasterMode(TIMER_MASTER_MODE_UPDATE);
  Timer1.setPeriod(500);
  Timer1.setMode(TIMER_CH2, TIMER_OUTPUT_COMPARE);
  Timer1.setCompare(TIMER_CH2, 1);
  Timer1.attachInterrupt(TIMER_CH2, blinkLED);
  Timer1.refresh();
  Timer1.resume();

  myADC1.calibrate();
  myADC1.setPins(pins, 2);

  myADC1.setSampleRate(ADC_SMPR_1_5);
  myADC1.setScanMode();

}

void myDMA() {
  dma1_ch1_Active = 1;

  myADC1.setDMA(dataPoints, 3000, (DMA_MINC_MODE | DMA_TRNS_CMPLT), DMA1_CH1_Event);
  myADC1.setTrigger(ADC_EXT_EV_TIM3_TRGO);
  myADC1.startConversion();
  while (dma1_ch1_Active >= 1) {};
  dma_disable(DMA1, DMA_CH1);
}

static void DMA1_CH1_Event() {
  dma1_ch1_Active = 0;
}

char read_char() {
  while (!Serial.available()) {

  };
  return Serial.read();
}

unsigned  char buf[6];
void loop() {

  unsigned char c;

  //tx(255);
  //tx(255);

  digitalWrite(PC13, LOW);
  char cc = read_char();
  digitalWrite(PC13, HIGH);

  if (cc == 0xE ) {
    buf[0] = cc;
    buf[1] = read_char();
    buf[2] = read_char();
    buf[3] = read_char();

  } else if ( cc == 0xF) {
    buf[0] = cc;

  } else {
    buf[0] = 0;
  }

if (buf[0] == 0xF) { // request do triger

    dataready = 0;
    myDMA();
    dataready = 1;
    tx(0);

  } else if (buf[0] == 0xE) {
    c = 0;

    if (dataready == 1) {
      // digitalWrite(PC13, LOW);
      if (buf[1] == 0x1) {
        delayMicroseconds(10);
        tx(0);
        tx(0);

        for (int i = 0; i < 1500; i++) {
          c = (unsigned char)map(dataPoints[i], 0, 1023, 1, 254)  ;
          if (c == 0) c = 1;
          tx(c);
        }
        tx(0);
      } else if (buf[1] == 0x2) {
        delayMicroseconds(10);
        tx(0);
        tx(0);
        for (int i = 1500; i < 3000; i++) {
          c =  (unsigned char)map(dataPoints[i], 0, 1023, 1, 254)  ;
          if (c == 0) c = 1;
          tx(c);

        }
        tx(0);
      } else {
        tx(0);
      }

      //digitalWrite(PC13, HIGH);

    } else {
      tx(255);
    }

  } else {
    tx(255);
  }

}

stevstrong, 2023-11-03:
Sorry, but I cannot understand what your code is doing.
How do you detect that the output is swapped?
Please simplify your sketch to a minimum in order to demonstrate the issue.

stevstrong, 2024-01-01:
Closing this due to inactivity from OP.

closed 2024-01-01 by stevstrong