rogerclarkmelbourne/Arduino_STM32 #903: Arduino IDE 2 Support

arslan437, 2023-03-18:
I have created a hardware folder in the document/Arduino/hardware and pasted the file downloaded from this repo and pasted in that folder. I can see the boards listed in the IDE. But there is that I am getting when try to compile 

```
exec: "/bin/arm-none-eabi-g++": file does not exist

Compilation error: exec: "/bin/arm-none-eabi-g++": file does not exist
```

rogerclarkmelbourne, 2023-03-18:
Arduino 2.0 not supported

Also read . 

https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

You did not install the GCC compiler by installing any othere ARM board from the boards manager

closed 2023-03-18 by rogerclarkmelbourne

arslan437, 2023-03-18:
Now I have followed the instruction and I am to compile the code with Arduino 2.0.4

arslan437, 2023-03-18:
@rogerclarkmelbourne  I got this error when I try to upload the code 

```
"C:\Users\Arslan\Documents\Arduino\hardware\Arduino_STM32/tools/win/maple_upload.bat" COM8 2 1EAF:0003 "C:\Users\Arslan\AppData\Local\Temp\arduino\sketches\6FFE11AFBECE6F3ADD2923C849955E8D/Blink.ino.bin" "{runtime.ide.path}"
'java' is not recognized as an internal or external command,
operable program or batch file.
```

rogerclarkmelbourne, 2023-03-18:
Post to https://www.stm32duino.com