rogerclarkmelbourne/Arduino_STM32 #909: How can I make 'STM32F103C8T6' DIY? help me.

SeongJongKwak, 2023-06-18:
hello. I am the my first time person dealing with 'STM32F103C8T6' this time.

There are 4 things I'm curious about.

Currently, I have purchased and used the 'Blue Pill' board.

1. I want to purchase the chip 'STM32F103C8T6' and make 'DIY, PCB' myself.
Q1) If so, how should I boot(bootloader)?

2. How to do a soft reset?
Q2) I am curious about the source code.

3. I want to test in conjunction with 'STM32F103C8T6' and 'DS3231'. By any chance, can I use the firmware as it is?
Q3) hardware I2C : PB_9-SAD, PB_8-SCL
Q4) Do you have any examples or methods?

4. How to use 'EEPROM.h' library?

stevstrong, 2023-06-18:
Please address these questions on
www.stm32duino.com.
Here should only bugs be reported.

closed 2023-06-18 by stevstrong

fpistm, 2023-06-19:
FYI already post on the forum: https://www.stm32duino.com/viewtopic.php?t=2013