rogerclarkmelbourne/Arduino_STM32 #911: Support for Arduino IDE 2.х

techmike78, 2023-08-28:
Good afternoon.
Is there support for Arduino IDE 2.х?

After installing this IDE and setting the dependencies and installing the board, there is no STM32 in the list of processors.
my task is to revive the project under STM32F103C8T6

[image]
[image]
[image]
[image]

rogerclarkmelbourne, 2023-08-28:
This repo does not contain a boards package.

fralbo, 2023-09-16:
I don't know if it's related to this, but I also tried to import this library to IDE2 and I get an INTERNAL ERRO 13, library not valid.
My goal is to try to use https://github.com/ag88/stm32duino_rtcadj to adjust the RTC on blackpill board.
Is there a solution please?

Cheers Alain

stevstrong, 2023-11-06:
As far as I understood, the poster claims that he cannot find the board i the boards manager.
I have shown the solution how a board can be selected.

He did not stated that there would be any problem with Java.
If so, then he should open a new issue (if there is not already one having this topic).

So for now I will close this issue as the original problem has been solved.
It can be reopened if there is still something not working as expected.

closed 2023-11-06 by stevstrong