rogerclarkmelbourne/Arduino_STM32 #914: Arduino IDE "Access Denied" on Uploading Sketch

dapostol73, 2023-12-12:
I followed this [tutorial](https://www.electroniclinic.com/stm32-bootloader-stm32f103c-programming-using-usb-cable/) on getting the Blue Pill setup, everything went fine until I tried to upload the sketch and got this.

`Sketch uses 12804 bytes (19%) of program storage space. Maximum is 65536 bytes.
Global variables use 2064 bytes (10%) of dynamic memory, leaving 18416 bytes for local variables. Maximum is 20480 bytes.
Access is denied.`

I'm runtime Windows 11 with the latest IDE and installed the Drivers using install_driver.bat from Arduino_STM32 repository.  I check my drives in the Device Manager and they say they are using 'Maple Serial (COM7)' however in the IDE it just say 'COM7' (not sure if this an issue)

[image]

Also, if I connect with Serial Monitor, the board seems to spam these lines repeatedly, which makes me assume the bootloader is working properly.

> Congratulations, you have installed the STM32duino bootloader
> See https://github.com/rogerclarkmelbourne/STM32duino-bootloader
> 
> For more information about Arduino on STM32
> See https://www.stm32duino.com

Please let me know if there is something I'm missing or an issue running this on Windows 11?

stevstrong, 2023-12-12:
I don't know how WIn11 manages the ports but from your post I think that Arduino access to the port is prohibited.
Have you installed the serial driver with admin right?
Try to start Arduino with admin rights.

board707, 2023-12-12:
@dapostol73 
If you talked about IDE 2.x, take the note that you must to close a Serial monitor before uploading the code to the board.

dapostol73, 2023-12-12:
Hi @board707 Yes, the Serial Monitor is close when uploading. I just used it to confirm that the bootloader was up loaded successfully and is working.