rogerclarkmelbourne/Arduino_STM32 #920: dfu upload failed on macOS + PlatformIO because cannot reset upload port

ohdarling, 2024-03-31:
When use STM32duino-bootloader + platformio + dfu upload method, upload will failed because upload-reset cannot work with the random serial port.

platformio.ini config:

```ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
board_build.core = maple
upload_protocol = dfu
build_flags = -DSERIAL_USB -DGENERIC_BOOTLOADER
```

Use PlatformIO upload command, will get logs below:

```
Compiling .pio/build/genericSTM32F103C8/FrameworkArduino/wirish_time.cpp.o
Compiling .pio/build/genericSTM32F103C8/src/main.cpp.o
Linking .pio/build/genericSTM32F103C8/firmware.elf
Checking size .pio/build/genericSTM32F103C8/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.1% (used 4320 bytes from 20480 bytes)
Flash: [===       ]  29.8% (used 19536 bytes from 65536 bytes)
Building .pio/build/genericSTM32F103C8/firmware.bin
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: /dev/cu.Bluetooth-Incoming-Port
Uploading .pio/build/genericSTM32F103C8/firmware.bin
*** [upload] Error 1
```

Because platformio will auto detect a serial port and pass to dfu upload script, in logs above it is `/dev/cu.Bluetooth-Incoming-Port`, but the serial port does not supprot reset operation, then the dfu upload script will failed.

It seems that upload-reset is not necessary on macOS, when I comment out the reset line, the uploading works.

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/tools/macosx/maple_upload#L34

```bash
${DIR}/upload-reset ${dummy_port_fullpath} 750
```

Then the upload was successful.

```
Checking size .pio/build/genericSTM32F103C8/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.1% (used 4320 bytes from 20480 bytes)
Flash: [===       ]  29.8% (used 19536 bytes from 65536 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: /dev/cu.Bluetooth-Incoming-Port
Uploading .pio/build/genericSTM32F103C8/firmware.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!

Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device

Download        [                         ]   0%            0 bytes
Download        [=                        ]   4%         1024 bytes
Download        [==                       ]   9%         2048 bytes
Download        [===                      ]  14%         3072 bytes
Download        [====                     ]  19%         4096 bytes
Download        [======                   ]  24%         5120 bytes
Download        [=======                  ]  29%         6144 bytes
Download        [========                 ]  34%         7168 bytes
Download        [=========                ]  39%         8192 bytes
Download        [===========              ]  44%         9216 bytes
Download        [============             ]  49%        10240 bytes
Download        [=============            ]  54%        11264 bytes
Download        [==============           ]  59%        12288 bytes
Download        [================         ]  64%        13312 bytes
Download        [=================        ]  69%        14336 bytes
Download        [==================       ]  74%        15360 bytes
Download        [===================      ]  79%        16384 bytes
Download        [=====================    ]  84%        17408 bytes
Download        [======================   ]  89%        18432 bytes
Download        [=======================  ]  94%        19456 bytes
Download        [======================== ]  99%        19536 bytes
Download        [=========================] 100%        19536 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
Waiting for /dev/cu.Bluetooth-Incoming-Port serial...Done
```

Currently, I am commenting it out to make the stm32duino-bootloader work. I am not sure if this line of code is useful in other scenarios, so I am opening an issue to request your team to handle it. Thank you.

stevstrong, 2024-03-31:
First: we do not support PlatformIO directly.

Second: I do not have any Mac, I cannot test it, so I cannot guarantee that your change does not break something else working with Arduino IDE.

Did you test it with Arduino IDE?

ohdarling, 2024-03-31:
I tested in Arduino IDE with following settings:

[image]

It seems reset failed does not break the whole dfu progress:

[image]

But it reported a error after dfu finished.

I think is it could add an parameter or environment to ignore upload-reset to `maple_upload` script, then platformio could add that parameter under macOS.

Thanks.

fpistm, 2024-04-01:
You didn't use the maple core with Arduino IDE while you use it for PIO.
So use the correct core following this wiki: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

ohdarling, 2024-04-01:
> You didn't use the maple core with Arduino IDE while you use it for PIO. So use the correct core following this wiki: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

Maple core was specified in platformio.ini:

```ini
board_build.core = maple
```

Reference:

https://docs.platformio.org/en/stable/platforms/ststm32.html#switching-between-arduino-cores

fpistm, 2024-04-01:
That is what I said. You use maple core with pio but not with arduino ide (based on your screen shot) 😉