rogerclarkmelbourne/Arduino_STM32 #923: ‘MCU basic current’ increases after PWM operation. help me.

SeongJongKwak, 2024-04-24:
‘MCU basic current’ increases after PWM operation.

- core: https://github.com/rogerclarkmelbourne/Arduino_STM32

1. If there is no operation, the total current of 0.07A flows.

2. Turn on PWM
- (source code)pinMode(PA6, PWM); pwmWrite(PA6, 5000);

3. Turn off PWM
- (source code)pinMode(PA6, PWM); pwmWrite(PA6, 0);

4. The total current has increased.
- A total current of 0.22A flows.
- MCU power consumption has increased.

5. I wanted to restore it to its original state through soft reset, but the current remains the same.
- (source code) nvic_sys_reset();
- Total current 0.22A

6. If you turn off the entire power and turn it back on,
- Total current 0.07A

The conclusion is that after using 'pwmWrite', the power consumption of the MCU increases (increases).
Even if the pwm is turned off, the ‘MCU power consumption’ remains increased.
Moreover, even with ‘software power reset’, ‘MCU power consumption’ has increased.

Q1. I don't know why.
Q2. I want to solve it.
- I think it needs to be powered down and then turned on again. Is there a way?

video
[https://drive.google.com/file/d/1O5L-9e ... sp=sharing](https://drive.google.com/file/d/1O5L-9eQFcDbqi5UzAxCit1_HdzUlec3J/view?usp=sharing)

help me.

board707, 2024-04-24:
Why do you call `pinMode(PWM)` again in step 3?
Try to switch the pin to default INPUT mode:
```
 pwmWrite(PA6, 0);
 pinMode(PA6, INPUT); 
```

fpistm, 2024-04-24:
For ref: https://www.stm32duino.com/viewtopic.php?t=2376

SeongJongKwak, 2024-04-24:
In the actual code, three PWMs are used.

const int PWM_P[] = {PB0, PA6, PA8};

   Timer1.pause(); Timer2.pause(); Timer3.pause(); Timer1.setChannel1Mode(TIMER_PWM); Timer2.setChannel2Mode(TIMER_PWM); Timer3.setChannel3Mode(TIMER_PWM);
   Timer1.setPeriod(2000); Timer2.setPeriod(2000); Timer3.setPeriod(2000);
   Timer1.resume(); Timer2.resume(); Timer3.resume();

board707, 2024-04-24:
Once you used 

`for (int i=0; i<3; i++){ pinMode(PWM_P,PWM); }`

you don't need all this:

> Timer1.pause(); Timer2.pause(); Timer3.pause(); Timer1.setChannel1Mode(TIMER_PWM); Timer2.setChannel2Mode(TIMER_PWM); Timer3.setChannel3Mode(TIMER_PWM);
> Timer1.setPeriod(2000); Timer2.setPeriod(2000); Timer3.setPeriod(2000);
> Timer1.resume(); Timer2.resume(); Timer3.resume();

SeongJongKwak, 2024-04-24:
1. The code is there.
for (int i=0; i<3; i++){ pinMode(PWM_P[i],PWM); }

2. Even if you omit the source code below, the result is the same.
Timer1.pause(); Timer2.pause(); Timer3.pause(); Timer1.setChannel1Mode(TIMER_PWM); Timer2.setChannel2Mode(TIMER_PWM); Timer3.setChannel3Mode(TIMER_PWM);
Timer1.setPeriod(2000); Timer2.setPeriod(2000); Timer3.setPeriod(2000);
Timer1.resume(); Timer2.resume(); Timer3.resume();

board707, 2024-04-24:
Sorry, but it seems to me that you are very new in all of this?
Please show your full schematic diagram and your code.

SeongJongKwak, 2024-04-24:
https://drive.google.com/file/d/1Zzc4RWhHiwDr26UmPr_283Ro_7VEi0S8/view

@board707
@stevstrong

1. The video was filmed in detail.
2. stevstrong You are right. But to be more precise,
The summed current is the corresponding power consumption because it is two regulators and an MCU. So, the video is not wrong. The entire 0.22A is not the MCU increase. However, as the MCU current increased, the regulator current also increased.
(The regulator is 5V, 3.3V.)
Now, the reason the current is reduced compared to the previous video is because the input voltage of the regulator is lowered.
  - And, the actual MCU is very hot.
  - So, I installed a heat sink.

3. When I tested it,
  pwmWrite(PA6, 3500); If you use , the MCU seems to get hot. What does it feel like?
  1) Basic mode
  2) Acceleration mode
  It operates in acceleration mode.
  However, what is interesting is that the ‘acceleration mode’ does not change by reinstalling the firmware or resetting the software. It only operates in 'default mode' when the power is turned on again.

SeongJongKwak, 2024-04-24:
My assumptions are as follows:

1. As pwmWrite is used, the current in the ‘MCU’ increases.
  - in terms
  1) Basic mode
  2) Acceleration mode
  Let’s say there are two modes.
  - When pwmWrite is used, it enters Acceleration mode.
  - Even if pwmWrite is set to '0', it goes into Acceleration mode.
  - As a result, the MCU current increases.

2. Through ‘firmware upload’ or ‘software reset’
You cannot escape ‘Acceleration mode’.

Here's what I feel right now:

board707, 2024-04-24:
> * the actual MCU is very hot.

It looks like your MCU was already fried before.

> * Basic mode
> 
> * Acceleration mode

What are these modes as applied to the STM32 core? Correct me, but I don't remember anything similar in the package under discussion.

stevstrong, 2024-04-24:
Try to deconnect everything from the MCU and measure the consumption again.

board707, 2024-04-24:
@SeongJongKwak 
If others wrote to you that they do not have this effect, then the problem is not in the stm32 package. This is probably due to a bug in your code or circuitry.

Once I will at home, I will test the consumption of the PWM on my STM32F103 and F411 boards.

SeongJongKwak, 2024-04-24:
@stevstrong @board707

After testing, it seems to be a hardware problem.

https://drive.google.com/file/d/1fPifqyLMslOs9-gaKDpF41FWIYI-lI2q/view?usp=sharing

I will send you a link to the circuit diagram.

Looking at U2, U3, and U4,
I connected PB0, PA6, and PA8 directly. Could this be a problem?

Anyway, in this circuit
1) After using PWM, power consumption increased.
2) The power consumption of the 3.3VDC regulator increased.
3) And the MCU was hotter than usual.
4) Through ‘firmware upload’ or ‘software reset’, it was equally hot.
5) It doesn't get hot until you turn the power back on.

board707, 2024-04-24:
What are the U2 U3 U4 ? 
IR2101 or something like?

SeongJongKwak, 2024-04-24:
@board707 
IR2104  Chip.

SeongJongKwak, 2024-04-24:
https://drive.google.com/file/d/182Xc9KkL8o6yt2_IVpiuzqAqqB9iLZhm/view

SeongJongKwak, 2024-04-24:
@board707 @SeongJongKwak @fpistm 

The big difference between 'soft reset' and 'hard reset' is 'RAM initialization'.
Is it possible to ‘initialize RAM’ through software?
Currently, it appears to be the difference between 'soft reset' and 'hard reset'.

board707, 2024-04-24:
@SeongJongKwak 
Why do you need a reset, if you found that the issue it was a hardware fault? 
You have to fix the fault rather than invent the workaround.

board707, 2024-04-24:
> The STM32 device should never get hot.

Indeed.
But if you damaged it already - it is a very common that the fried mcu would very hot.

SeongJongKwak, 2024-04-24:
@board707 @stevstrong

You are right. And thank you for participating in this debate.

I found the cause. (Additional circuit diagram)
url: https://drive.google.com/file/d/1sUG4Njh6N8fxbmLm5A1JxwvhqjxqI6ch/view?usp=sharing

- MR1: Shunt resistance
- L4,L5: Bead resistance (0 ohm~1 ohm)
- Directly connected to ADC to measure current.

After removing L4 and L5, the problem disappeared. It seems that the leakage current between the MCU and ADC pin is the cause.

Anyway, I found the cause, and I sincerely thank you for your kind response.

SeongJongKwak, 2024-04-24:
There is no problem with the library at all. Literally, it was a circuit problem.
Thank you very much for your help in resolving the issue.

SeongJongKwak, 2024-04-27:
@board707 @stevstrong @fpistm 

https://youtu.be/zdrVvyuxAjQ

It was handled well. Success.
Thank you.

closed 2024-04-27 by stevstrong