rogerclarkmelbourne/Arduino_STM32 #927: Is 'STM32F103C8' 'digitalWrite' and 'digitalRead' possible in FAST mode?

SeongJongKwak, 2024-05-17:
Is 'STM32F103C8' 'digitalWrite' and 'digitalRead' possible in FAST mode?

[Same article, forum]
https://www.stm32duino.com/viewtopic.php?t=2401

hello.
[Status]
1. MCU: STM32F103C8
2. CORE1: https://github.com/rogerclarkmelbourne/Arduino_STM32

In the case of Arduino (AVR), there are the following libraries.
[https://www.arduinolibraries.info/libra ... write-fast](https://www.arduinolibraries.info/libraries/digital-write-fast)
'digitalWrite' Operates 'digitalRead' quickly.

Is it possible to operate ‘Fast mode’ through ‘CORE1’?
If anyone knows, please help.

board707, 2024-05-17:
> Is it possible to operate ‘Fast mode’ through ‘CORE1’?

Hi
You can easily write such "fast mode" itself, using the direct port manipulation:
```
int pin = PA5;
uint32_t* pin_set_register = portSetRegister(pin);
uint32_t pin_set_mask = digitalPinToBitMask(pin);

// digitalWrite() HIGH
*pin_set_register = pin_set_mask;

// digitalWrite() LOW
*pin_set_register = pin_set_mask << 16;
```

SeongJongKwak, 2024-05-17:
@board707 

Thank you.
I understand how to do it.

Is there a method using 'digitalRead'?

board707, 2024-05-17:
I think that the syntax seems to be the similar, but I never need a "fast" digitalRead() in my projects.

stevstrong, 2024-05-17:
use:
uint32_t* pin_get_register = portSetRegister(pin);
uint16_t value = portInputRegister(pin_get_register)  & digitalPinToBitMask(pin);

SeongJongKwak, 2024-05-17:
@stevstrong 
code : 
int pin7 = PA5;
uint16_t value1 = *portInputRegister(pin7)& digitalPinToBitMask(pin7);

error : 
\generic_stm32f103c/variant.h:7:44: error: base operand of '->' is not a pointer
 #define portInputRegister(port)    ( &(port->regs->IDR) )

SeongJongKwak, 2024-05-17:
code : 
uint16_t value1 = *portInputRegister(digitalPinToPort(pin7)) & digitalPinToBitMask(pin7);

Thank you.

SeongJongKwak, 2024-05-17:
@stevstrong

It doesn't work.

code:
const int SIG[] = {PA5, PA4, PA0, PA15};
   //------------------------------------------------ --------------------//
   value1 = *portInputRegister(digitalPinToPort(SIG[0])) & digitalPinToBitMask(SIG[0]);
   value2 = *portInputRegister(digitalPinToPort(SIG[1])) & digitalPinToBitMask(SIG[1]);
   value3 = *portInputRegister(digitalPinToPort(SIG[2])) & digitalPinToBitMask(SIG[2]);
   uint16_t hall_y = 0; uint16_t hall_b = 0; uint16_t hall_g = 0;
   if(value1==0){ hall_y = 0; }else{ hall_y = 1; }
   if(value2==0){ hall_b = 0; }else{ hall_b = 1; }
   if(value3==0){ hall_g = 0; }else{ hall_g = 1; }
   //------------------------------------------------ --------------------//

stevstrong, 2024-05-17:
What exactly do you mean "it doesn't work"? Which part does not work?

board707, 2024-05-17:
> It doesn't work.

The port access syntax is correct, the error somewhere else in your code.