rogerclarkmelbourne/Arduino_STM32 #929: Solution to use PB3 for things other than JTAG (Proposal for documentation note)

leecyrille, 2024-05-18:
I found I could not use the PB3 pin for INPUT_PULLUP mode with an STM32F103C or STM32F103V, until I disabled JTAG functionality. If anyone wants to use PB3 for other functions, I has success by adding these lines to my setup first:
```
void setup() {

//Add these lines to free up PB3 
__IO uint32_t *AFIO_MAPR = (__IO uint32_t *)0x40010004;  //Use direct register access to disable JTAG and keep SWD enabled
 *AFIO_MAPR |= 0x02000000; //Set the JTAG-DP disable bit, SW-DP enable bit

//The rest of your setup
pinMode(PB3, INPUT_PULLUP);

}
```

Maybe this is something to add in documentation for people to search later? I assume changing the default behavior would break JTAG for other people?

board707, 2024-05-19:
The solution is simpler
In order to use JTAG pins (PB3, PB4, PA15), you just need to add a command to the setup:
```
enableDebugPorts();
```
Please note that this is only necessary when using st-link for upload; when using the Serial,  pins are enabled by default

leecyrille, 2024-05-19:
> The solution is simpler In order to use JTAG pins (PB3, PB4, PA15), you just need to add a command to the setup:
> 
> ```
> enableDebugPorts();
> ```
> 
> Please note that this is only necessary when using st-link for upload; when using the Serial, pins are enabled by default

This makes more sense, thank you!

closed 2024-05-19 by leecyrille