rogerclarkmelbourne/Arduino_STM32 #932: Unable to upload program when using STM32duino BootLoader

TTXL3099, 2024-06-06:
Hello, thank you very much for developing this.I had problems using it on my Windows10 computer. For both arduinoIDE 1.8.19 and 2.0.0, he couldn't use the stm32duino bootloarder uploader. It seems that this is not a hardware problem, but a system or software problem.I have tried Windows10PRO 22H2 (19045) and Windows10 ltsc2019(17763). It can compile normally, but cannot be uploaded using the stm32duino BootLoader. Uploading with ST-LINK is no problem. From the output, the problem appears to be coming from "jSSC" in "JAVA".
[image]
[image]
[image]
[image]
[image]

### Below are several logs mentioned in the output.

The first came from LTSC 2019 running in VMware Workstation Pro, and the remaining four came from my computer win10 22H2.
I tried installing jre-8u411 on Windows, but it didn't work.

[hs_err_pid6644.log](https://github.com/user-attachments/files/15668207/hs_err_pid6644.log)
[hs_err_pid13024.log](https://github.com/user-attachments/files/15668213/hs_err_pid13024.log)
[hs_err_pid21312.log](https://github.com/user-attachments/files/15668218/hs_err_pid21312.log)
[hs_err_pid22196.log](https://github.com/user-attachments/files/15668220/hs_err_pid22196.log)
[hs_err_pid10836.log](https://github.com/user-attachments/files/15668225/hs_err_pid10836.log)

All information output from the arduinoIDE 2.0.0 uploa
[output form IDE2.0.0.txt](https://github.com/user-@stevstrong achments/files/15669601/output.form.IDE2.0.0.txt)
d process. @stevstrong @stevstrong2 @rogerclarkmelbourne Can you help me?

ag88, 2024-08-15:
this issue is followed up and resolved in this thread
https://www.stm32duino.com/viewtopic.php?p=1477#p14707

apparently, the JSSC DLL errors in windows can be resolved by installing an updated copy of JSSC from the github repository, replacing that delivered with the core
https://github.com/java-native/jssc
but that the user needs to figure out and manually replace it in the installed core folders.

fpistm, 2024-08-15:
That's why I removed java dependency in the stm32 core. Moreover it's solved issue  when Arduino IDE 2 is used as it is not java dependent.😁

closed 2024-08-15 by TTXL3099

TTXL3099, 2024-08-15:
> 此问题已在此线程中得到跟进和解决 https://www.stm32duino.com/viewtopic.php?p=14701#p14701
> 
> 显然，可以通过从 github 存储库安装 JSSC 的更新副本来解决 Windows 中的 JSSC DLL 错误，替换核心 https://github.com/java-native/jssc提供的副本 ，但用户需要弄清楚并在已安装的核心文件夹中手动替换它。

I'm not sure if I should create a pull request to fix this. Replacing JSSC2.8 with JSSC2.9.6 is the solution to this problem.
@stevstrong Maybe your repository needs to do this too to avoid problems?